erlerobot/gym-gazebo #188: Turtlebot Example-gzclient not displayed

adilmahsan, 2018-11-24:
Hey guys,

I'm having a bit of a problem. I created a test setup using the instructions from issue#148 and issue#90 for Python2.7 on Ubuntu 16.04.

I'm trying to run the turtlebot example.

The example runs, the learning training is being executed, and i'm getting the plot, but i don't see the graphics. I tried starting gzclient from a new terminal, but i don't see anything and it goes back to the next line after a few seconds. 

I would appreciate any help in the matter. I have copied the terminal status and the terminal log.

Thank you. :)

[image]

The log is as follows:
```
user@osboxes:~/gym-gazebo/gym_gazebo/envs/installation$ bash turtlebot_setup.bash
user@osboxes:~/gym-gazebo/gym_gazebo/envs/installation$ cd 
user@osboxes:~ cd gym-gazebo/examples/turtlebot 
user@osboxes:~/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py
`ROS_MASTER_URI=http://localhost:10001

GAZEBO_MASTER_URI=http://localhost:10002

... logging to /home/user/.ros/log/7b97230c-ef7a-11e8-81a9-080027d6939f/roslaunch-osboxes-3507.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Roscore launched!
Unable to register with master node [http://localhost:10001]: master may not be running yet. Will keep trying.
started roslaunch server http://osboxes:46415/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [3530]
Gazebo launched!
ROS_MASTER_URI=http://osboxes:10001/

setting /run_id to 7b97230c-ef7a-11e8-81a9-080027d6939f
process[rosout-1]: started with pid [3562]
started core service [/rosout]
... logging to /home/user/.ros/log/7b97230c-ef7a-11e8-81a9-080027d6939f/roslaunch-osboxes-3548.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

/usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:62: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1 and perhaps the rcParam as well
  'version 2.1 and perhaps the rcParam as well')
/usr/lib/python2.7/dist-packages/matplotlib/backend_tools.py:78: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1, and some tools might change name
  'version 2.1, and some tools might change name')
started roslaunch server http://osboxes:36797/
![turtlebot_visualisation_problem](https://user-images.githubusercontent.com/27216313/48962821-d8988b80-ef53-11e8-8e1d-263d9a06b0d3.png)

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/user/gym-ga...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

ROS_MASTER_URI=http://osboxes:10001/

process[gazebo-1]: started with pid [3755]
process[spawn_turtlebot_model-2]: started with pid [3759]
process[mobile_base_nodelet_manager-3]: started with pid [3761]
process[cmd_vel_mux-4]: started with pid [3762]
process[bumper2pointcloud-5]: started with pid [3763]
process[robot_state_publisher-6]: started with pid [3764]
[ INFO] [1543017030.503898729]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1543017030.758416912]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1543017036.168160825, 0.680000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1543017037.041727160, 31.940000000]: Physics dynamic reconfigure ready.
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
[ INFO] [1543017046.904967081, 210.660000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1543017047.078324073, 210.660000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1543017048.263321940, 210.660000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1543017048.264434691, 210.660000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1543017048.333807557, 210.660000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1543017048.428025725, 210.660000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1543017048.428362109, 210.660000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1543017048.448357891, 210.660000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1543017048.448882034, 210.660000000]: Will publish tf. [mobile_base]
[ INFO] [1543017048.464906892, 210.660000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1543017048.468446739, 210.660000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1543017048.490602086, 210.660000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1543017048.530941725, 210.660000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[spawn_turtlebot_model-2] process has finished cleanly
log file: /home/user/.ros/log/7b97230c-ef7a-11e8-81a9-080027d6939f/spawn_turtlebot_model-2*.log
[ INFO] [1543017048.561245343, 210.660000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1543017048.570013718, 210.660000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1543017048.582786888, 210.660000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1543017048.591244015, 210.660000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1543017048.591406881, 210.660000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
/usr/lib/python2.7/dist-packages/matplotlib/backend_bases.py:2437: MatplotlibDeprecationWarning: Using default event loop until function specific to this GUI is implemented
  warnings.warn(str, mplDeprecation)
EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -163     Time: 0:00:42
EP: 2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -160     Time: 0:00:45
```

nzlz, 2018-11-24:
I believe this is a gazebo installation related issue. Try running `gzclient --verbose`, maybe you could see some extra output. You could also try reinstalling gazebo... but before that take a look at [this](http://answers.gazebosim.org/question/18724/gazebo-9-gui-doesnt-launch/?answer=20050#post-id-20050).

Also know that you have installation instructions available [here](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md).

adilmahsan, 2018-11-25:
Thank you for your reply and the suggestions... I ran it with verbose, it keeps waiting for the server for 30 seconds and then times out.... Yes I think you're right, it seems like an installation issue... I'll try following the standard installation procedure... It was a bit ambiguous with regards to certain details regarding the installation of the pre-requisites so i went with the other one... 

Thanks again... :)

adilmahsan, 2018-11-28:
Hey people,

So i tried again with the standard installation instructions for V16.04. And i'm still facing the same problem, Gazebo visualization won't start. The training starts though and i do see the plot. 

The response with gzclient --verbose was as follows:

```
gzclient --verboseuser@osboxes:~/git/gym-gazebo/examples/turtlebot$ gzclient --verbose
Gazebo multi-robot simulator, version 8.6.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Err] [ConnectionManager.cc:119] Failed to connect to master in 30 seconds.
[Err] [gazebo_shared.cc:83] Unable to initialize transport.
[Err] [gazebo_client.cc:62] Unable to setup Gazebo
user@osboxes:~/git/gym-gazebo/examples/turtlebot$ 
```
The script execution yielded the following result:

```
user@osboxes:~/git/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py
ROS_MASTER_URI=http://localhost:13959

GAZEBO_MASTER_URI=http://localhost:13960

... logging to /home/user/.ros/log/dd22ade0-f348-11e8-9537-080027b202ad/roslaunch-osboxes-3158.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Roscore launched!
Unable to register with master node [http://localhost:13959]: master may not be running yet. Will keep trying.
started roslaunch server http://osboxes:45645/
ros_comm version 1.12.14

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES

auto-starting new master
process[master]: started with pid [3186]
ROS_MASTER_URI=http://osboxes:13959/

setting /run_id to dd22ade0-f348-11e8-9537-080027b202ad
Gazebo launched!
/usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:62: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1 and perhaps the rcParam as well
  'version 2.1 and perhaps the rcParam as well')
/usr/lib/python2.7/dist-packages/matplotlib/backend_tools.py:78: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1, and some tools might change name
  'version 2.1, and some tools might change name')
... logging to /home/user/.ros/log/dd22ade0-f348-11e8-9537-080027b202ad/roslaunch-osboxes-3204.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

process[rosout-1]: started with pid [3220]
started core service [/rosout]
started roslaunch server http://osboxes:32959/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
[... 27 lines cut ...]
[ INFO] [1543435511.535957159, 0.010000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1543435511.863276264, 0.010000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1543435511.863482240, 0.010000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1543435511.923875351, 0.010000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1543435512.171709247, 0.010000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1543435512.172456730, 0.010000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1543435512.185570226, 0.010000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1543435512.186155859, 0.010000000]: Will publish tf. [mobile_base]
[ INFO] [1543435512.198617021, 0.010000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1543435512.207708407, 0.010000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1543435512.247142922, 0.010000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1543435512.285652524, 0.010000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1543435512.317744760, 0.010000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1543435512.320195350, 0.010000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1543435512.326912648, 0.010000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[spawn_turtlebot_model-2] process has finished cleanly
log file: /home/user/.ros/log/dd22ade0-f348-11e8-9537-080027b202ad/spawn_turtlebot_model-2*.log
[ INFO] [1543435512.335957864, 0.010000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1543435512.336266127, 0.010000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
[ INFO] [1543435512.352016951, 0.060000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1543435513.184153341, 2.210000000]: Physics dynamic reconfigure ready.
/usr/lib/python2.7/dist-packages/matplotlib/backend_bases.py:2437: MatplotlibDeprecationWarning: Using default event loop until function specific to this GUI is implemented
  warnings.warn(str, mplDeprecation)
EP: 1 - [alpha: 0.2 - gamma: 0.
```

adilmahsan, 2018-11-28:
Also, I had a few queries regarding some ambiguities I had with regards to the installation instructions:

**Regarding Installation Procedure for V16.04:**

OpenAI Gym: Is it supposed to be installed for Python2 or Python3?

After the insallation part, there's the usage part. In the usage part, it is mentioned that we should use sudo pip install -e . Should it be run with pip or pip3? 

I assume the final script has to be run with python (and not python3).

Could you please clarify the architecture a little bit with regards to the standard application procedure with regards to what parts are supposed to be handled by python 2 and which by python 3.

**Regarding Installation Procedure for Ubuntu 18.04:**

In this one, it seems that we are handling almost everything with python2. Is that correct? But we were still installing three python 3 packages: python3-vcstool python3-pyqt4 python3-defusedxml. Is there any particular reason for this?

Also, I tried the 18.04 installation but there were some files in gazebo_melodic.repos which didn't have a melodic version and they would generate an error during compilation. Could you please tell me how we can avoid these issues? Like I tried substituting to another branch but it didn't work.

nzlz, 2018-11-29:
I am sorry but I can't help you with the gazebo client issue. It could be a gazebo version issue since you are using 8.6.0 and the installation instructions were made using 8.1.1, but I really don't know.

> Regarding Installation Procedure for V16.04:

Regarding the installation.. the main script, `whatever_qlean.py`, must be executed using python2, python3 simply does not work. Same for `pip install -e .`, version 2 works, version 3 I don't know.

> Regarding Installation Procedure for Ubuntu 18.04:

Those python3 libraries where just used to satisfy requirements related to the package compilation inside the catkin_workspace. I am sure there is a better solution for this, but is this not important (I believe).

About the non-melodic repos, I did not have any issue related to the version of those when compiling the workspace. Which is the issue you are having?

_Note that gym-gazebo is maintained by the community, so feel free to make your own contributions with fixes or new code :)_

nzlz, 2018-11-29:
Hi @adilmahsan . 
Ubuntu 18 installation updated: https://github.com/erlerobot/gym-gazebo/pull/190

adilmahsan, 2018-11-29:
Ah cool... Thanks... :) 

I'll try again... :)

adilmahsan, 2018-11-29:
@nzlz  thanks man.. so i tried the ubuntu 18 VM with the updates... and it compiled correctly.. :)

But there's a thing, i'm still not getting the visualization which makes me wonder if i'm running it correctly... 

what i do is, i run the test example on a particular terminal, then once it has been executed and it's in the process of learning episodes, i open a new terminal and i run the command gzclinet --verbose... should i be doing something else/additionally or is that it?

nzlz, 2018-11-29:
Oh, i didn't know you were using a virtual machine. This issue is related to this kind of environment if the problem is the same in Ubuntu 16 and 18. Tell me about the specific characteristics of your setup and I will try to help you.

About the gzclient, you are right. You can see [here ](https://youtu.be/8hxCBkgp95k?t=14) or [here ](https://youtu.be/WRePKygK5p0?t=62)that is simply loads the visuals of gazebo after executing it.

adilmahsan, 2018-11-29:
thanks @nzlz... :) 

I just downloaded the virtual box versions for 18.04 and 16.04 and ran them with the default settings with 2 GB ram using Oracle Virtual Box...

the virtual boxes are available here:

https://www.osboxes.org/ubuntu/

the software is available here:

https://www.virtualbox.org/

nzlz, 2018-11-30:
The first this you have to know is that solving problems using Reinforcement Learning requires long time, as many iterations have to be completed until some kind of learning is reached. This is why I recommend you install Ubuntu in your system and dont use a VM. 

But if you want to keep it the VM way.. try deactivating the _3D Hardware accelatation_ option in the settings of your machine in virtualbox. 

[image]

adilmahsan, 2018-12-01:
Thanks for your help man... :) 

I wanted to try a repeatable and functional installation procedure before installing it directly on the system... 

The 3D acceleration setting was already disabled... But I think you are probably right that it might be an issue related to the virtual environment...

I will give it a shot with a system and hope for the best... :)

Thanks again... :)

HU-Yi-ming, 2018-12-02:
Hi,
i'm encounter with the same problem as adil ahsan, i follow the instruction of installation for python3 #90 and python2 #148,but it is unhelp, separately  type gzserver and gzclient is worked with no error in my machine, my work enviornement is ros-kinetic v1.12.14 and gazebo 8.6.
Thanks.

HU-Yi-ming, 2018-12-02:
Hi,
i'm encounter with the same problem as adilmahsan, i follow the instruction of installation for python3 #90 and python2 #148,but it is unhelp, separately  type gzserver and gzclient is worked with no error in my machine, my work enviornement is ros-kinetic v1.12.14 and gazebo 8.6.
Thanks.

nzlz, 2018-12-02:
Hi @HU-Yi-ming ,
You should provide more information about your issue. Also try to follow [this ](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md)installation instructions.
If you are using a VM, I suggest you abandon this idea, at least for the moment.

HU-Yi-ming, 2018-12-02:
Thank you very much for repley @nzlz 

I'm not using VM, rather installed ubuntu directly on my computer. I do follow the install instruction that you referred and #90 and #148 ,addittionly tryed the install using pip2 and pip3.but I also encounted same problem as adilmahsan 

my compile enviroment is:
ubuntu 16.04,
 kinetic /rosversion: 1.12.14
gazebo 8.6
pip2+python2.7

but when it come to type gzclient --verbose,  gazebo is not showing. but I just typed gzserver and gzclient is worked.
[image]

The log is as follows:

> ros@ros:~/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py 
> /usr/local/lib/python2.7/dist-packages/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
>   result = entry_point.load(False)
> ROS_MASTER_URI=http://localhost:13539
> 
> GAZEBO_MASTER_URI=http://localhost:13540
> 
> ... logging to /home/ros/.ros/log/268b3312-f623-11e8-9b08-b8763fc5403f/roslaunch-ros-18409.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ros:44747/
> ros_comm version 1.12.14
> 
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
> 
> auto-starting new master
> process[master]: started with pid [18419]
> ROS_MASTER_URI=http://ros:13539/
> 
> setting /run_id to 268b3312-f623-11e8-9b08-b8763fc5403f
> Roscore launched!
> Gazebo launched!
> process[rosout-1]: started with pid [18450]
> started core service [/rosout]
> /usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:62: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1 and perhaps the rcParam as well
>   'version 2.1 and perhaps the rcParam as well')
> /usr/lib/python2.7/dist-packages/matplotlib/backend_tools.py:78: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1, and some tools might change name
>   'version 2.1, and some tools might change name')
> ... logging to /home/ros/.ros/log/268b3312-f623-11e8-9b08-b8763fc5403f/roslaunch-ros-18446.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://ros:45733/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /bumper2pointcloud/pointcloud_radius: 0.24
>  * /cmd_vel_mux/yaml_cfg_file: /home/ros/gym-gaz...
>  * /robot_description: <?xml version="1....
>  * /robot_state_publisher/publish_frequency: 30.0
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
[... 24 lines cut ...]
> log file: /home/ros/.ros/log/268b3312-f623-11e8-9b08-b8763fc5403f/spawn_turtlebot_model-2*.log
> [ INFO] [1543749152.177510747, 9.780000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
> [ INFO] [1543749152.192229443, 9.780000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
> [ INFO] [1543749152.350962751, 9.780000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
> [ INFO] [1543749152.351119099, 9.780000000]: Starting Laser Plugin (ns = /)!
> [ INFO] [1543749152.355311129, 9.780000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
> [ INFO] [1543749152.459843145, 9.780000000]: Starting plugin Kobuki(ns = //)!
> [ WARN] [1543749152.460004710, 9.780000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
> [ INFO] [1543749152.461735880, 9.780000000]: Kobuki(ns = //): <tf_prefix> = 
> [ INFO] [1543749152.462282960, 9.780000000]: Will publish tf. [mobile_base]
> [ INFO] [1543749152.466075173, 9.780000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
> [ INFO] [1543749152.467462080, 9.780000000]: Kobuki(ns = //): Advertise Odometry[odom]!
> [ INFO] [1543749152.479073377, 9.780000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
> [ INFO] [1543749152.486556565, 9.780000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
> [ INFO] [1543749152.494270358, 9.780000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
> [ INFO] [1543749152.495628540, 9.780000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
> [ INFO] [1543749152.497155221, 9.780000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
> [ INFO] [1543749152.498781451, 9.780000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
> [ INFO] [1543749152.498876463, 9.780000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
> /usr/lib/python2.7/dist-packages/matplotlib/backend_bases.py:2437: MatplotlibDeprecationWarning: Using default event loop until function specific to this GUI is implemented
>   warnings.warn(str, mplDeprecation)
> EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -98     Time: 0:00:06
> EP:2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -172     Time: 0:00:06

`

nzlz, 2018-12-02:
@HU-Yi-ming could you check that after setting `<arg name="gui" default="true"/>` in the [GazeboCircuit2TurtlebotLidar_v0](https://github.com/erlerobot/gym-gazebo/blob/ad99c9189da04514b707c1b0a37cb5a2308d6c09/gym_gazebo/envs/assets/launch/GazeboCircuit2TurtlebotLidar_v0.launch#L6), the GUI is still not working? This will try to launch the gzclient automatically. If it does not work, send me the error message please.

Anyway, I will take a look at the Ubuntu 16.04 installation since it seems to be causing problems.

HU-Yi-ming, 2018-12-03:
thanks @nzlz 
What surprised me was that when I modified gui arguments to be true, gazebo gui appeared.But this seems to mean that I have to set this up in the future.
really thankful.
[image]

adilmahsan, 2018-12-03:
Thanks @nzlz... It's also working for me now on the virtual machines... :)

nzlz, 2018-12-16:
Same thing happening to me now in a fresh melodic installation, don't know the reason yet. 
Needs research/fix asap.

nzlz, 2018-12-29:
**Solution**:

In order to launch the `gzclient` and be able to connect it to the running `gzserver`:
1. Open a new terminal.
2. Source the corresponding setup script, which will update the _GAZEBO_MODEL_PATH_ variable: e.g. `source setup_turtlebot.bash`
3. Export the _GAZEBO_MASTER_URI_, provided by the [gazebo_env](https://github.com/erlerobot/gym-gazebo/blob/7c63c16532f0d8b9acf73663ba7a53f248021453/gym_gazebo/envs/gazebo_env.py#L33). You will see that variable printed at the beginning of every script execution. e.g. `export GAZEBO_MASTER_URI=http://localhost:13853`

**Note**: This instructions are needed now since `gazebo_env` creates a random port for the GAZEBO_MASTER_URI, which allows to run multiple instances of the simulation at the same time.

closed 2018-12-29 by nzlz

HareshMiriyala, 2019-03-02:
tip : use ctrl+s to pause the process from printing continuously on screen, ctrl+q to resume printing