erlerobot/gym-gazebo #189: Installation error during "bash setup_melodic.bash"

wimboyuniarto, 2018-11-29:
Hi 
I tried to install on Ubuntu 18.04, when i tried to install using "bash setup_melodic.bash, I get the following error:
[ 56%] Building CXX object ar_track_alvar/CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::ProjPoints::AddPointsUsingChessboard(IplImage*, double, int, int, bool)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:54:26: error: ‘cvFindChessboardCorners’ was not declared in this scope
  int pattern_was_found = cvFindChessboardCorners(gray, cvSize(etalon_rows, etalon_columns), corners, &point_count);
                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:73:3: error: ‘cvDrawChessboardCorners’ was not declared in this scope
   cvDrawChessboardCorners(image, cvSize(etalon_rows, etalon_columns),
   ^~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::Calibrate(alvar::ProjPoints&)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:365:29: error: ‘CV_CALIB_USE_INTRINSIC_GUESS’ was not declared in this scope
   &calib_K, &calib_D, 0, 0, CV_CALIB_USE_INTRINSIC_GUESS);
                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:363:2: error: ‘cvCalibrateCamera2’ was not declared in this scope
  cvCalibrateCamera2(object_points, image_points, &point_counts,
  ^~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:363:2: note: suggested alternative: ‘Calibrate’
  cvCalibrateCamera2(object_points, image_points, &point_counts,
  ^~~~~~~~~~~~~~~~~~
  Calibrate
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::CalcExteriorOrientation(std::vector<CvPoint3D64f>&, std::vector<CvPoint2D64f>&, alvar::Pose*)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:656:2: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(object_points, image_points, &calib_K, NULL, &ext_rodriques_mat, &ext_translate_mat);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::CalcExteriorOrientation(std::vector<CvPoint3D64f>&, std::vector<alvar::Point<CvPoint2D64f> >&, CvMat*, CvMat*)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:692:2: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(&world_mat, &image_mat, &calib_K, &calib_D, rodriques, tra);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::Camera::CalcExteriorOrientation(const CvMat*, CvMat*, CvMat*, CvMat*)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:727:2: error: ‘cvFindExtrinsicCameraParams2’ was not declared in this scope
  cvFindExtrinsicCameraParams2(object_points, image_points, &calib_K, &calib_D, rodriques, tra);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(std::vector<CvPoint3D64f>&, alvar::Pose*, std::vector<CvPoint2D64f>&) const’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:756:2: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:756:2: note: suggested alternative: ‘ProjectPoints’
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
  ProjectPoints
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(const CvMat*, const CvMat*, const CvMat*, CvMat*) const’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:769:2: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, rotation_vector, translation_vector, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:769:2: note: suggested alternative: ‘ProjectPoints’
  cvProjectPoints2(object_points, rotation_vector, translation_vector, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
  ProjectPoints
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Camera::ProjectPoints(const CvMat*, const alvar::Pose*, CvMat*) const’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:780:2: error: ‘cvProjectPoints2’ was not declared in this scope
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:780:2: note: suggested alternative: ‘ProjectPoints’
  cvProjectPoints2(object_points, &ext_rodriques_mat, &ext_translate_mat, &calib_K, &calib_D, image_points);
  ^~~~~~~~~~~~~~~~
  ProjectPoints
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp: In member function ‘void alvar::Homography::Find(const std::vector<alvar::Point<CvPoint2D64f> >&, const std::vector<alvar::Point<CvPoint2D64f> >&)’:
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:861:2: error: ‘cvFindHomography’ was not declared in this scope
  cvFindHomography(&src_pts, &dst_pts, &H);
  ^~~~~~~~~~~~~~~~
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/src/Camera.cpp:861:2: note: suggested alternative: ‘Homography’
  cvFindHomography(&src_pts, &dst_pts, &H);
  ^~~~~~~~~~~~~~~~
  Homography
ar_track_alvar/CMakeFiles/ar_track_alvar.dir/build.make:62: recipe for target 'ar_track_alvar/CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o' failed
make[2]: *** [ar_track_alvar/CMakeFiles/ar_track_alvar.dir/src/Camera.cpp.o] Error 1
CMakeFiles/Makefile2:37105: recipe for target 'ar_track_alvar/CMakeFiles/ar_track_alvar.dir/all' failed
make[1]: *** [ar_track_alvar/CMakeFiles/ar_track_alvar.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j 1" failed
## ROS workspace compiled ##

Can you help please to fix this error?
Thanks

nzlz, 2018-11-29:
My bad! please in the `gazebo_ros_melodic.repos` file replace : 
 ```
 ar_track_alvar:
    type: git
    url: https://github.com/sniekum/ar_track_alvar
    version: melodic-devel
  ar_track_alvar_msgs:
    type: git
    url: https://github.com/sniekum/ar_track_alvar_msgs
    version: melodic-devel`
 ```
with this:
 ```
  ar_track_alvar:
    type: git
    url: https://github.com/ros-perception/ar_track_alvar
    version: melodic-devel`
 ```

I will create a new PR once I test this from scratch!

wimboyuniarto, 2018-11-29:
Thanks a lot..
I have been solved that but i get error on [ 68%] Building CXX object kobuki_desktop/kobuki_gazebo_plugins/CMakeFiles/gazebo_ros_kobuki.dir/src/gazebo_ros_kobuki.cpp.o
/home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/kobuki_desktop/kobuki_gazebo_plugins/src/gazebo_ros_kobuki.cpp:7:10: fatal error: gazebo/math/gzmath.hh: No such file or directory
 #include <gazebo/math/gzmath.hh>

nzlz, 2018-11-29:
If you go to` /home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/kobuki_desktop`. Can you confirm that the remote url and the branch of that repository are: 
```
git remote -v -> url: https://github.com/yujinrobot/kobuki_desktop
git branch    -> branch: devel
```

wimboyuniarto, 2018-11-29:
I find this one 
     - git:
    local-name: kobuki_desktop
    uri: 'https://github.com/yujinrobot/kobuki_desktop.git'
    version: indigo

nzlz, 2018-11-29:
Strange that you have the `indigo `branch if you followed the instructions. Anyway, just checkout to the `devel `branch and try to `catkin_make` again.

```
git checkout -b devel
cd /home/wimbo/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/catkin_ws
catkin_make
```

adilmahsan, 2018-11-29:
Thanks @nzlz ... it compiled without any problems on the ubuntu 18 machine... :)

wimboyuniarto, 2018-11-29:
thank you very much @nzlz , finally there was no error, yeah i missed some instruction

closed 2018-12-29 by nzlz