erlerobot/gym-gazebo #191: Installation error during "bash setup_melodic.bash" with sophus

benbenji, 2018-12-05:
Hi.when I run setup_melodic.bash,it output can not find Sophus package in ecl_linear_algebra cmake files.Then I found that melodic version ecl_linear_algebra need "Sophus" but kinetic versions need "sophus". so I try to install the melodic version of Sophus instead of kinetic version but it still has this problem:
 CMake Error at ecl_core/ecl_linear_algebra/CMakeLists.txt:29 (get_target_property):
  get_target_property() called with non-existent target "Sophus::Sophus".
 did any gays meet this problem ? I will appreciate it very much if you can give me a solution

nzlz, 2018-12-05:
Make sure you install sophus the correct way, follow these [instructions](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#install-sophus). 

You are right about the version, you could try using the `release/1.0.1-melodic` branch. But this is not the issue here, you just need to install sophus.

benbenji, 2018-12-06:
I solve that question thank you very much!.but I have another question after successfully complie the whole ros workspace ,when I run the circuit_turtlebot_lidar_qlearn.py
the output is show as : 
catkin_ws1/devel/lib/python3/dist-packages/gazebo_msgs/srv/_SpawnModel.py", line 103, in serialize
    _x = _x.encode('utf-8')
AttributeError: 'bytes' object has no attribute 'encode'
By analyzing the code, I found that this is related to the python version. because the  wrong code is:
 if python3 or type(_x) == unicode:
        _x = _x.encode('utf-8')
        length = len(_x)
      buff.write(struct.pack('<I%ss'%length, length, _x))
so I want to ask which python version should be used to compile and run melodic version of gym_gazebo?

nzlz, 2018-12-06:
The main script needs to be executen in Python2. 
Please check the [issues](https://github.com/erlerobot/gym-gazebo/issues) first if you encounter new problems, it is very likely that the solution has been posted before.

closed 2018-12-29 by nzlz