erlerobot/gym-gazebo #192: ROS2 migration.

nzlz, 2018-12-12:
Hello everyone,

I will be trying to port gym-gazebo to **ROS2** the following days/weeks. I just wanted to share this here, just in case someone is willing to offer some advice/help. Since the structure is going to be updated, suggestions regarding a new configuration are welcomed too.

I will be working in my own fork for now: [nzlz/gym-gazebo](https://github.com/nzlz/gym-gazebo). And I will be updating the state of the project in the [wiki page](https://github.com/nzlz/gym-gazebo/wiki/Current-state-of-the-ROS1-to-ROS2-migration.) of the repo.

My first task will be to try to reproduce the `pause/unpause_physics` and `reset_simulation` service calls. Leaving the pause/unpause for later versions, when its officially implemented is an option too.

Please feel free to share your thoughts.

vmayoral, 2018-12-12:
I think this is great. Can we maybe raise a new thread at https://discourse.ros.org sharing this same thing and asking for additional contributors?

nzlz, 2018-12-13:
@vmayoral done! Pending aproval.

I want to arrive at some conclusion regarding the necessity or not of implementing `pause`/`unpause`/`reset` in the ROS2 version. First two are not officially available yet.
@rkojcev @YueErro I have a few questions regarding the MARA envs/algorithms, maybe you guys can help me out:

- Should I be focusing on the `real_mara_top_3dof_ros2_v0.py` env? If available, could you provide a really simple example script to test that env? 
- I see you do not use services provided by gazebo_ros_pkgs (`pause`/`unpause`/`reset`) in these envs. Is the reason simply that you found a more convenient way to solve this, or did these kind of services suppose any problem during the training?

Next is just a bit of curiosity, note that my knowledge in machile learning is very limited:
- Related to previous one: is it good practice that the robot still moving outside of the `step()` function? Why is it done this way?
- Related to previous one: the reward is captured at the beginning of the `step()` function and the action to be taken is executed at the end. I just want to confirm that this is handled correctly in this case because the action to be taken is related to the new position of the robot, which is a deterministic movement I believe. What do you think about algorithms like the ones tested with the Turtlebot, where the action is performed in a short range of time, not exact. Should this be avoided or is it okay to handle it with the initial /`unpause` and posterior /`pause` of physics in gazebo?

Any kind of suggestions are welcomed. Sorry for long post and thanks in advance.

YueErro, 2018-12-13:
Hi @nzlz, 
>- Should I be focusing on the `real_mara_top_3dof_ros2_v0.py` env? If available, could you provide a really simple example script to test that env? 

I have not personally worked on `real_mara_top_3dof_ros2_v0.py`, we are currently working on the reward shaping in order to avoid collisions, and some randomization features in [gazebo_mara_top_orient_collision_v0.py](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py). I can provide you the training script to test this collision env.

>- I see you do not use services provided by gazebo_ros_pkgs (`pause`/`unpause`/`reset`) in these envs. Is the reason simply that you found a more convenient way to solve this, or did these kind of services suppose any problem during the training?

We are not using gazebo_ros_pkgs and we make use of our own [reset fuction](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L856-L886) when [done](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L826) is true after each training step.
I guess in `real_mara_top_3dof_ros2_v0.py` is quite similar.

>- Related to previous one: is it good practice that the robot still moving outside of the `step()` function? Why is it done this way?

The robot does not move outside the `step()` function.

> - Related to previous one: the reward is captured at the beginning of the `step()` function and the action to be taken is executed at the end. I just want to confirm that this is handled correctly [...]

Let's say the initial pose of the robot is _ob0_, first action _ac0_, new pose after executing _ac0_ is _ob1_ and second action _ac1_:
It is in _ob0_ and when it does the step, you get the reward from _ob0_ and it executes _ac0_ and it gets to _ob1_.
In the next step, it is in _ob1_, you get the reward from _ob1_, it executes _ac1_  and it gets to _ob2_.
And so on...

In short, the reward value is regarding the previous action:

You can have a look into [this](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L761-L832), here we get the reward from the action executed at that step.

rkojcev, 2018-12-13:
Hi @nzlz,

Thanks for having a look and 

> Should I be focusing on the real_mara_top_3dof_ros2_v0.py env? If available, could you provide a really simple example script to test that env?
@YueErro  will provide you a script and help you start how to use it.

> I see you do not use services provided by gazebo_ros_pkgs (pause/unpause/reset) in these envs. Is the reason simply that you found a more convenient way to solve this, or did these kind of services suppose any problem during the training?

Actually those calls are causing problems and I would avoid using them, actually we do not need to use them and they are designed and tested for discrete action space.

> Related to previous one: is it good practice that the robot still moving outside of the step() function? Why is it done this way?

The robot is not moving outside the step function, can you explain me where you found this piece of code? After step is done the obs, reward, etc are returned to the algorithm. It might be a bug we need to have a look at if something is going on after the step function.

> Related to previous one: the reward is captured at the beginning of the step() function and the action to be taken is executed at the end. I just want to confirm that this is handled correctly in [...]
This is not quite true, the actions are in continuous space now and we are using  algorithms, such as PPO, to utilize observations in continuous space. Robot movements in general and robotics deals with contentious space actions, therefore we had to look at algorithms which provide us the possibility to use continuous space instead of discrete space.

>  What do you think about algorithms like the ones tested with the Turtlebot, where the action is performed in a short range of time, not exact. Should this be avoided or is it okay to handle it with the initial /unpause and posterior /pause of physics in gazebo?
Unfortunately Q-learning does not perform well on our robots, I think you should have a look at **PPO**, TRPO or ACKTR for example, or even A2C for continuous actions. I would reccomend trying things out with PPO. We have couple of example environments, that we managed to converge using PPO and MARA

I would suggest you to take a look at the following script: as @YueErro suggested which is the most complex one we have, where we also utilize some functions such as collisions: https://github.com/erlerobot/gym-gazebo/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py

More simple ones: https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_3dof_v0.py

I would suggest you also as the first step is to convert all the ROS1 topics to ROS2 topics in the environment itself. 
We can also give you access to our internal repository regarding our RL infrastructure so you can have a look the things we have over there.

nzlz, 2018-12-13:
Thanks both! @YueErro @rkojcev 

Regarding the `step` function, I was talking about [real_mara_top_3dof_ros2_v0](https://github.com/erlerobot/gym-gazebo/blob/ad99c9189da04514b707c1b0a37cb5a2308d6c09/gym_gazebo/envs/MARA/real_mara_top_3dof_ros2_v0.py#L486). I guess this a test version or something, so I will just stick to the ones you referenced.

I will take a quick look at the algorithms you mentioned, which I'm not familiar with. Also start from migrating topics.

rkojcev, 2018-12-13:
Yes that is true @nzlz we use that one to launch the NN trained in gym_gazebo in the real robot. We actually do not use that script for training, but can be good inspiration how to use ROS2 topics and add them in gym_gazebo

nzlz, 2019-01-13:
@rkojcev @YueErro I'm migrating [gazebo_mara_top_orient_vision_collision_v0.py](https://github.com/erlerobot/gym-gazebo/blob/14dfe24903a32626fc6938e18a209f79370d437f/gym_gazebo/envs/MARA/gazebo_mara_top_orient_vision_collision_v0.py#L752) to ROS2 and I notice again the unusual `step` function. We already had a similar conversation about this in this same issue, so I just want to know if i should update/fix the structure of the `step` function in this environment or it is supposed to work this way.

What i mean is, instead of the usual step structure:
- action
- observation
- reward
- done (status)

In this particular Env the structure is:
- reward
- done (status)
- action
- observation

YueErro, 2019-01-13:
I would say to make it consistent in all the environments.

I guess the difference between these step structures is not going to change the training results. I think is just a matter of taking the start pose (in our case when all the motors are set to 0, when MARA is "up" ) as the first reward and leaving the last action without reward, or doing the opposite.

In case we realize one of those structures is better than the other one, we can easily fix it by changing the order.

It is true that in all the papers i have read they are using the second structure:
- reward
- done (status)
-  action
- observation

I wonder why and if there is actually a relevant reason.

nzlz, 2019-01-13:
Hmm really curious about this.. i really think we should avoid second option since it mixes information from different timesteps (in a conceptual way). 

I think the end result it just the same because the obsevation is not updated until the next action, a "lucky" bug. Will reseach a bit about this tomorrow.

rkojcev, 2019-01-13:
The order changes the results significantly. The reason behind is that in PPO1 we also take prev action into account. This is not done in PPO2 for example.

The vision code is not maintained for a while now and is done as proof of concept to see if we can train if we have constant feeding(dynamic) 3D point location.

I would recommend to make it consistent for now, since we are neglecting PPO1 and we will need to evaluate once again the order of things with certain algorithms. We should make it in the following order: 
- action
- observation
- reward
- done (status)

The papers are little bit old and yes in that cases and in ppo1 they used some weird reversed order, but I think we can switch to more logical order now and test it.

YueErro, 2019-01-13:
For me fisrt option makes more sense.

In fact, this [collision detection](https://github.com/erlerobot/gym-gazebo/blob/14dfe24903a32626fc6938e18a209f79370d437f/gym_gazebo/envs/MARA/gazebo_mara_top_orient_vision_collision_v0.py#L819-L835) is wrong.
We are suppose to get the reward of the previous action, but if it collides after executing the new action we are getting the reward from this new action instead of the previous one.

rkojcev, 2019-01-13:
@YueErro lets discuss this tomorrow. If we collide the reward needs to be bad and updated as bad since it collides. we do not want to have good reward when it collides so it needs to be updated appropriately depending on the case

rkojcev, 2019-01-14:
@nzlz for the order of the things in the step function, can you make it consistent everywhere such as:

- [action](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L744)
- [observation](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L748-L750)
- (reward)[https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L752-L755]
- [adjusted reward when collision occurs](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L766)
- [no collision reward](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L791-L806)
- [done (status)](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L808)

Please note that rewards are negative up till the point when they sitisfy certain distance condition, in our case we set that to 0.005 meters (5 mm), [code](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L794)

nzlz, 2019-01-14:
Already updated in the ROS2 envs :+1: , will check later again in detail.

vmayoral, 2019-01-15:
Thanks for sharing this wish list @mkhansen-intel,

> - Can this be refactored to use AI Gym as a library instead of being a fork of AI Gym?
>   - This will allow you to more easily stay updated rather than having to rebase

Connected to https://github.com/erlerobot/gym-gazebo/issues/121#issuecomment-454540449. This sounds like a good improvement @nzlz.

>  * Can the examples be refactored to share code instead of having code duplication, as I suggested in my issue #134?

+1. We should be able to refactor things now that's being coded again.

>  * Can you add support for other robot arms such as the UR5 or Franka Panda, which are both used by MoveIt!
> * Could it provide ROS Actions output so that it could be compatible with other robot arms? If the output was compatible with the [ros_control](http://wiki.ros.org/ros_control) package it could be used with many robots. This is analogous to using /cmd_vel topic to command a robot base such as a turtlebot.

We did this in the past. Created extensions to support other robots or specific interfaces and maintenance became a burden thereby, for gym-gazebo2, we'll try a different path. We'll aim towards supporting a vendor-agnostic and interoperable interface for robots and robot modules: the [Hardware Robot Information Model (HRIM)](https://github.com/erlerobot/hrim). A simple example of such interface is the 3D camera/depth sensor model: https://acutronicrobotics.com/docs/technology/hrim/model/models/sensor/3d%20camera (@mkhansen-intel I'd recommend having a look at it, it'll be extremely helpful if the Intel's components, e.g. the RealSense camera would comply with this interface as well).

_Note that HRIM aims to inspire itself from the de-facto ROS messages, including those used by `ros_control` thereby, essentially, what you're asking will be realized_.