erlerobot/gym-gazebo #194: Clock synch between ROS and Gazebo

pulver22, 2018-12-14:
Hi,

I was experimenti way to speed up the simulation using the real time factor of Gazebo. But I noticed that even if Gazebo is faster to implement the action he received, the learning script is not fast enough because it's listening on the Wall Clock. Therefore we are only really speeding up the simulation as we could. 

Investigating the problem I noticed that, even if we set `use_sim_time` to `True` in order to force the synchronization between Ros and Gazebo, the simulator needs to be the first node to be launch in order to reset the Clock to 0.

The solution is actually easy as I implemented on my own version [here](https://github.com/pulver22/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_env.py).

In `gazebo_env.py`, we just have to comment out the line launching the roscore. It doesn't make sense to launch a roscore when we are using a launchfile for Gazebo (which will initialise a roscore itself). Plus this cause the issue related to the clock.

My code implements also a debugging code to check which clock ROS is using.

I hope I was clear but I can explain better if required. I wanted to open a pull request but I don't know how to do it for just one file without adding all my personal change.

nzlz, 2018-12-14:
Hi @pulver22 ,
Thanks for spending time in gym-gazebo! I guess repo mantainers will take a look at your solution in the PR. If you want to make a PR with just one file you could do something like this (not tested!):
- Create a new branch: `git checkout -b clock_synch`
- Add a new remote : `git remote add erlerobotics https://github.com/erlerobot/gym-gazebo.git`
- Get state of erlerobotics: `git fetch erlerobotics`
- Apply changes and discard local: `git reset --hard erlerobotics/master`
- Change gazebo_env file with your solution + commit + create the PR from clock_synch branch.

rkojcev, 2018-12-14:
@pulver22 as @nzlz suggested it would be great if you make a PR and we will review your solution.

closed 2018-12-29 by nzlz