erlerobot/gym-gazebo #199: Gazebo not launching

C-monC, 2019-01-07:
Hello. I'm trying to run the example for qlearn (the first example after installing according to the instructions in INSTALL.md).

My problem is that Gazebo is not launching, but the example seems to be running without problems (it updates the rewards and draws the graph of rewards using matplotlib). Calling gzclient does not open any windows / simulations.

# Commands (in order)
## Terminal 1
    cd gym-gazebo/gym_gazebo/envs/installation/
    bash turtlebot_setup.bash

    cd gym-gazebo/examples/turtlebot
    python circuit_turtlebot_lidar_qlearn.py

## Terminal 2 (called after the last command in terminal 1)
    gzclient

# The graphed rewards
[image]

# Console Output
```ROS_MASTER_URI=http://localhost:14457

GAZEBO_MASTER_URI=http://localhost:14458

Gazebo launched!
Unable to register with master node [http://localhost:14457]: master may not be running yet. Will keep trying.
... logging to /home/c-mon/.ros/log/6b00a37e-128e-11e9-a9a6-54271eab675b/roslaunch-cmon-2505.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://cmon:33951/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/c-mon/gym-g...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [2532]
ROS_MASTER_URI=http://cmon:14457/

setting /run_id to 6b00a37e-128e-11e9-a9a6-54271eab675b
process[rosout-1]: started with pid [2560]
started core service [/rosout]
process[gazebo-2]: started with pid [2563]
process[spawn_turtlebot_model-3]: started with pid [2573]
process[mobile_base_nodelet_manager-4]: started with pid [2578]
process[cmd_vel_mux-5]: started with pid [2579]
process[bumper2pointcloud-6]: started with pid [2581]
process[robot_state_publisher-7]: started with pid [2582]
[ INFO] [1546873856.198608007]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1546873856.199736184]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1546873857.550105189, 0.480000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1546873858.222715682, 3.130000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1546873858.275494727, 3.130000000]: Physics dynamic reconfigure ready.
[ INFO] [1546873858.282414252, 3.130000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1546873858.399893003, 3.130000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1546873858.400007314, 3.130000000]: Starting Laser Plugin (ns = /)
[ INFO] [1546873858.593073846, 3.130000000]: Starting plugin Kobuki(ns = //)
[ WARN] [1546873858.593227714, 3.130000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[spawn_turtlebot_model-3] process has finished cleanly
log file: /home/c-mon/.ros/log/6b00a37e-128e-11e9-a9a6-54271eab675b/spawn_turtlebot_model-3*.log
[ INFO] [1546873858.815299614, 3.130000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1546873858.815391861, 3.130000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1546873858.815687958, 3.130000000]: Will publish tf. [mobile_base]
[ INFO] [1546873858.817565308, 3.130000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1546873858.818073929, 3.130000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1546873858.820376177, 3.130000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1546873858.822361821, 3.130000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1546873858.824862703, 3.130000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/velocity!
[ INFO] [1546873858.825425694, 3.130000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1546873858.826072686, 3.130000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1546873858.826703673, 3.130000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1546873858.826748795, 3.130000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -166
Time: 0:00:06
EP: 2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -145     Time: 0:00:07
EP: 3 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -137     Time: 0:00:08
EP: 4 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.89] - Reward: -139     Time: 0:00:09
EP: 5 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.89] - Reward: -131     Time: 0:00:10
EP: 6 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.89] - Reward: -174     Time: 0:00:10
EP: 7 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.89] - Reward: -171     Time: 0:00:10```

nzlz, 2019-01-07:
Solution was posted in [this issue](https://github.com/erlerobot/gym-gazebo/issues/188#issuecomment-450485080). 

You can also remove [these 2 ](https://github.com/erlerobot/gym-gazebo/blob/7c63c16532f0d8b9acf73663ba7a53f248021453/gym_gazebo/envs/gazebo_env.py#L27-L28)lines if you are not going to use multiple instances.

I will update the README.

C-monC, 2019-01-08:
Thanks it worked.

closed 2019-01-08 by C-monC