erlerobot/gym-gazebo #200: Error when running Qlearning example

Neuan, 2019-01-15:
Hello everybody!

We are having some issues to install the Gym-Gazebo in our 16.04 Ubuntu computer with ROS Kinetic. We have followed the instructions, but it does not work for us when we try to launch the qlearning example. 

This is the error that appears:

ERROR: cannot launch node of type [gazebo_ros/gzserver]: can't locate node [gzserver] in package [gazebo_ros]
ERROR: cannot launch node of type [gazebo_ros/spawn_model]: can't locate node [spawn_model] in package [gazebo_ros]

Thank you for the help!

nzlz, 2019-01-15:
Did you install gazebo? http://gazebosim.org/tutorials?tut=install_ubuntu

Neuan, 2019-01-15:
Yes I did. I have just updated Gazebo, and now another error appears:

ResourceNotFound: gazebo_ros
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/studerende/catkin_ws/src/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src
ROS path [2]=/opt/ros/kinetic/share

nzlz, 2019-01-15:
> ROS path [1]=/home/studerende/catkin_ws/src/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src

gym-gazebo should not be located inside a catkin_ws, I suggest you move it to your home directory.

But regarding your problem. Did you run the following command before executing the algorithm?:

```
cd gym_gazebo/envs/installation
bash setup_kinetic.bash		

```

Neuan, 2019-01-15:
I have taken the folder out of the workspace. I had also run that bash code, but it is still giving the same error as stated before of ResourceNotFound: gazebo_ros and also the next one:

/usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:62: UserWarning: Treat the new Tool classes introduced in v1.5 as expermiental for now, the API will likely change in version 2.1 and perhaps the rcParam as well
/usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:: UserWarning: Treat the new Tool classes introduced in v1.5 as expermiental for now, the API will likely change in version 2.1 and some tools might change name

nzlz, 2019-01-16:
Regarding the big issue, `gazebo_ros` not found. gazebo_ros is a package inside gazebo_ros_packages repository, which is installed inside the catkin_ws created for the gym. In order to make sure you install it properly:
1. Remove your current catkin_ws inside gym-gazebo (... `gym-gazebo/gym_gazebo/envs/installation/catkin_ws`)
2. Edit gazebo.repos and try replacing this line with `kinetic-devel`. Seems like you did not face compilation issues, but just in case.
3. Run `setup_kinetic.bash` and make sure the compilation does not fail.
4. Load q-learning correspondent `setup_turtlebot.bash`.