erlerobot/gym-gazebo #201: When I reinstalled numpy, I encountered this problem:No module named 'gym_gazebo'

Lv-xt, 2019-01-18:
I tried many methods but none of them worked.

nzlz, 2019-01-18:
You have to install gym-gazebo:
https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#install-gym-gazebo

Lv-xt, 2019-01-18:
> You have to install gym-gazebo:
> https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#install-gym-gazebo

I tried, but the problem still exists when I python circuit_turtlebot_lidar_qlearn.py

nzlz, 2019-01-18:
Make sure you install it using `pip`, not `pip3`. If you keep getting the same error, please provide extra information about your setup and the error trace.

Lv-xt, 2019-01-18:
> Make sure you install it using `pip`, not `pip3`. If you keep getting the same error, please provide extra information about your setup and the error trace.

I install it in  conda envs. before i reinstall numpy I hadn't met this problem.

nzlz, 2019-01-18:
What about adding the location of gym-gazebo to PYTHONPATH? Is it possible that you are messing python2 and 3 environments?

sayment, 2019-01-20:
It may related to #183

Lv-xt, 2019-01-21:
> What about adding the location of gym-gazebo to PYTHONPATH? Is it possible that you are messing python2 and 3 environments?

I reinstalled ubuntu, but it didn't work.But i find I have no gym_gazebo in pip list.how to solve it?thank you.

nzlz, 2019-01-21:
what is the output of `sudo pip install -e .` or `sudo -H pip install -e . ` as @sayment referenced. you should be able to install it that way once you are located in the gym-gazebo root directory.

nzlz, 2019-01-21:
I recommend you stay in Ubuntu18. Either way, just reinstall everything and be careful with python related installations not related with the gym. Good luck :+1:

Lv-xt, 2019-01-25:
> I recommend you stay in Ubuntu18. Either way, just reinstall everything and be careful with python related installations not related with the gym. Good luck +1

> > what is the output of `sudo pip install -e .` or `sudo -H pip install -e . ` as @sayment referenced. you should be able to install it that way once you are located in the gym-gazebo root directory.
> 
> > what is the output of `sudo pip install -e .` or `sudo -H pip install -e . ` as @sayment referenced. you should be able to install it that way once you are located in the gym-gazebo root directory.
> 
> I tried but it didn't work. I will reinstall ubuntu tomorrow and try i it again.If it still doesn't work
> .I try to install it in ubuntu16.Thanks very much

> I recommend you stay in Ubuntu18. Either way, just reinstall everything and be careful with python related installations not related with the gym. Good luck +1

I find i can import gym_gazebo in ~/gym-gazebo ,but can't import it in other location. How to solve it please?

nzlz, 2019-01-25:
When you install the package via pip3, it will create a link to the python .egg, locaten inside /gym-gazebo. So in `/usr/local/lib/python(your version)/dist-packages/ `you should find `gym-gazebo.egg-link` . Which basically means you can import if from anywhere.

What is your exact error, where/how do you try to import the module.

Lv-xt, 2019-01-25:
> What is your exact error, where/how do you try to import the module.

I import it by:
cd gym-gazebo
source activate aa
python
import gym_gazebo
MY error is in other location, i can't import it by  'import  gym_gazebo' and show: no module named "gym_gazebo"

Lv-xt, 2019-01-26:
> Hello @Lv-xt,
> 
> are you using conda environment to install all the packages? I'm not familiar with these kind of environment. If you're using the environment of conda, could you please do it outside of this environment?. I'll try to figure out your problem.

yes, I install all packages in it.
I successfully import it without a conda environment. 
Thanks a lot.

Lv-xt, 2019-01-26:
> When you install the package via pip3, it will create a link to the python .egg, locaten inside /gym-gazebo. So in `/usr/local/lib/python(your version)/dist-packages/ `you should find [...]

Thank you very very much! I solve the problem had been bothering me for a long time

closed 2019-01-26 by nzlz