erlerobot/gym-gazebo #202: ImportError: dynamic module does not define init function (init_tf2)

XuboLyu, 2019-01-21:
@rkojcev @vmayoral @ahcorde @imuguruza @LanderU 
**Robot**
Turtlebot

**Target**
Make the turtlebot move in a custom environment

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Ubuntu16.04
 - ROS version [e.g. ROS2 Crystal] ROS kinetic 
 - Gazebo version [e.g. Gazebo 9.6.0] Gazebo 8

I am using Python3 to run the programme. And I rebuild "tf" package for python3 via catkin build since original "tf" just work for python2.

Now I can load "tf" from the correct location, as you can see from error msg below. But I don't know how to fix the error:
```
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://cs-mars-01.cmpt.sfu.ca:34866/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/xlv/gym-gaz...
 * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 10
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    depthimage_to_laserscan (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    laserscan_nodelet_manager (nodelet/nodelet)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [23492]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 03c917f8-1dca-11e9-adea-b06ebfbb4fe5
process[rosout-1]: started with pid [23505]
started core service [/rosout]
process[gazebo-2]: started with pid [23526]
process[gazebo_gui-3]: started with pid [23533]
process[spawn_turtlebot_model-4]: started with pid [23537]
process[mobile_base_nodelet_manager-5]: started with pid [23538]
process[cmd_vel_mux-6]: started with pid [23540]
process[bumper2pointcloud-7]: started with pid [23543]
process[robot_state_publisher-8]: started with pid [23553]
process[laserscan_nodelet_manager-9]: started with pid [23572]
process[depthimage_to_laserscan-10]: started with pid [23624]
Traceback (most recent call last):
  File "/home/xlv/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model", line 32, in <module>
    import tf.transformations as tft
  File "/home/xlv/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 30, in <module>
  File "/home/xlv/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf2_ros/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
  File "/home/xlv/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/python3/dist-packages/tf2_py/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 38, in <module>
ImportError: dynamic module does not define init function (init_tf2)

```

nzlz, 2019-01-22:
I don't know how to help you here, you could try asking in `gazebo_ros_pkgs` since the error you are getting starts in an import inside that package. As always, I suggest you stick to python2.

XuboLyu, 2019-01-23:
Thanks for your reply @nzlz 

I've fixed it. It's basically due to the incompatibility between some ROS python2 package and python3 interpreter.

closed 2019-01-23 by XuboLyu