erlerobot/gym-gazebo #203: Custom Environment and Robot Implementation

sayment, 2019-01-23:
Hi All, I am master student at IMU. For my thesis, i would like to build custom environment and DJI Matrice M100 UAV in Gazebo and control it using ROS.
UAV will be this one: [https://github.com/caochao39/hku_m100_gazebo](link)
System informations as follows: 
Ubuntu 16.04, ROS Kinetic and Gazebo 8.6

My purpose is just to localize any random point on the open environment using reinforcement learning by UAV. Most likely it should be dqn. 

Could you help me to guide on how can i implement custom environment and M100 UAV above into gym-gazebo module?
I would like to know that what should i learn and what should i do step by step.

Regards.

nzlz, 2019-01-23:
> My purpose is just to localize any random point on the open environment using reinforcement learning by UAV. Most likely it should be dqn.

With localize you mean, go to a certain point right?

> I would like to know that what should i learn and what should i do step by step.

 Start with the Turtlebot. We have simple environments for it and implementations such as q-learn. You can read the [whitepaper](https://arxiv.org/abs/1608.05742) we published with the release, where those simple environments are explained.

Once you get confident in understanding the Environment - algorithm structure of gym-gazebo. The next step is to understand how each environment communicates with the robot though ROS. Calls to corresponding /topic /services etc..

Then you can start playing with DQN algorithm for the turtlebot or even other environments. You should get really into the depths of DQN is you want to achieve a training like the one you describe.

So yeah.. once you know how the structure of gym-gazebo you should be able just replicate it and adapt it for your own needs.

sayment, 2019-01-23:
> > My purpose is just to localize any random point on the open environment using reinforcement learning by UAV. Most likely it should be dqn.
> 
> With localize you mean, go to a certain point right?
> 
> > I would like to know that what should i learn and what should i do step by step.

Not exactly. I would like to make the robot find the point by trying. I will try some algorithms for finding the point. The robot will not know the position of the point.

> Start with the Turtlebot. We have simple environments for it and implementations such as q-learn. You can read the [whitepaper](https://arxiv.org/abs/1608.05742) we published with the release, where those simple environments are explained.
> 
> Once you get confident in understanding the Environment - algorithm structure of gym-gazebo. The next step is to understand how each environment communicates with the robot through ROS. Calls to corresponding /topic /services etc..
> 
> Then you can start playing with DQN algorithm for the turtlebot or even other environments. You should get really into the depths of DQN is you want to achieve a training like the one you describe.
> 
> So yeah.. once you know how the structure of gym-gazebo you should be able just replicate it and adapt it for your own needs.

I already read the paper and run the first qlearning example. It looks like if i change both environment and robot, i should write almost everything from scratch. And it's very complicated for now. It is better to work on working turtlebot and just changing the environment.
Thanks for your reply.