erlerobot/gym-gazebo #205: Gazebo simulation is missing

nameofuser1, 2019-01-27:
Hello!

I can not see the simulation via my gzclient.
What I did:
1. Run gzserver in one window.
2. Run gzclient in the second window.
3. Run example in the third window with `python circuit2_turtlebot_lidar_qlearn.py `

Here is output:
```
ROS_MASTER_URI=http://localhost:12505

GAZEBO_MASTER_URI=http://localhost:12506

Gazebo launched!
Unable to register with master node [http://localhost:12505]: master may not be running yet. Will keep trying.
... logging to /home/yuriy/.ros/log/1f50e462-2287-11e9-b089-0c54154d6af2/roslaunch-yuriy-OMEN-by-HP-Laptop-25714.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://yuriy-OMEN-by-HP-Laptop:40283/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/yuriy/gym-g...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [25740]
ROS_MASTER_URI=http://yuriy-OMEN-by-HP-Laptop:12505/

setting /run_id to 1f50e462-2287-11e9-b089-0c54154d6af2
process[rosout-1]: started with pid [25766]
started core service [/rosout]
process[gazebo-2]: started with pid [25773]
process[spawn_turtlebot_model-3]: started with pid [25780]
process[mobile_base_nodelet_manager-4]: started with pid [25781]
process[cmd_vel_mux-5]: started with pid [25782]
process[bumper2pointcloud-6]: started with pid [25783]
process[robot_state_publisher-7]: started with pid [25784]
[ INFO] [1548629937.288086128]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1548629937.289201374]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1548629938.199621395, 0.590000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1548629938.224662415, 9.670000000]: Physics dynamic reconfigure ready.
[ INFO] [1548629938.783080911, 5.220000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1548629938.788575677, 5.220000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1548629938.844790711, 5.220000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1548629938.844874743, 5.220000000]: Starting Laser Plugin (ns = /)
[ INFO] [1548629938.849074355, 5.220000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1548629938.952849293, 5.220000000]: Starting plugin Kobuki(ns = //)
[ WARN] [1548629938.952936484, 5.220000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1548629938.953753068, 5.220000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1548629938.954034984, 5.220000000]: Will publish tf. [mobile_base]
[ INFO] [1548629938.955776387, 5.220000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1548629938.956475676, 5.220000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1548629938.959222138, 5.220000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1548629938.961799718, 5.220000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1548629938.964792500, 5.220000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/velocity!
[ INFO] [1548629938.965665861, 5.220000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1548629938.966446363, 5.220000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1548629938.967456844, 5.220000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1548629938.967546515, 5.220000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
[spawn_turtlebot_model-3] process has finished cleanly
log file: /home/yuriy/.ros/log/1f50e462-2287-11e9-b089-0c54154d6af2/spawn_turtlebot_model-3*.log
EP: 1 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -153     Time: 0:00:03
EP: 2 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -148     Time: 0:00:04
EP: 3 - [alpha: 0.2 - gamma: 0.8 - epsilon: 0.9] - Reward: -156     Time: 0:00:05
```

Looks like model is spawned but I can not see it in my simulation.
Thanks!

nzlz, 2019-01-28:
Please follow these instructions: https://github.com/erlerobot/gym-gazebo#display-the-simulation . 

Note that you only need 2 terminals: one with the running script, which launches gzserver on its own, and another one with the gzclient.

nameofuser1, 2019-01-28:
> Please follow these instructions: https://github.com/erlerobot/gym-gazebo#display-the-simulation .
> 
> Note that you only need 2 terminals: one with the running script, which launches gzserver on its own, and another one with the gzclient.

Oh, sure. Thanks!

closed 2019-01-28 by nameofuser1