erlerobot/gym-gazebo #206: Turtlebot doesn't move in gym-gazebo/examples/turtle/circuit2_turtlebot_lidar_qlearn.py

cavinluo, 2019-01-30:
**Robot**
Turtlebot in circuit2_turtlebot_lidar_qlearn.py

**Environment/Algorithm**
Ubuntu 16.04+Python 2.7+ROS Kinetic

**Describe the bug**
When I try to run the examples in gym-gazebo/examples/turtle/circuit2_turtlebot_lidar_qlearn.py.
Roscore was successfully launched and the turtlebot also showed up with gzclient, but turtlebot doesn't move and there is no learning process show. 

I tried to solve the prolem following the Wrapper monitor error solution #20
Change the circuit2_turtlebot_lidar_qlearn.py
by replacing the line as:
#env = gym.wrappers.Monitor(env, outdir, force=True)
  env = gym.wrappers.Monitor(env, directory=outdir, force=True, write_upon_reset=True)

and keep
env._flush(force=True)

and I confirm thhat I run the example through python not python3 as shown in the following printscreen.
**Screenshots**
[image]

[image]

**Error trace**
There is false showing that "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  result = entry_point.load(False)"
and in display_plot.py, it showed that AttributeError: 'module' object has no attribute 'monitoring'

I tried the solutions from the  #20,  #113, #159, but the turtlebot stand there still.

Thank you very much for your help and kind advises.

nzlz, 2019-01-30:
Hi @cavinluo , thanks for you detailed explanation of the error.

Regarding the smallest problem first, that plotting utility has been in disuse for some time. One of the reasons is that a plot is already spawned with the simulation. Check https://github.com/erlerobot/gym-gazebo/blob/cf30f243dcc177de246d5f9debbabd2665c9b098/examples/turtlebot/circuit2_turtlebot_lidar_qlearn.py#L83 You are not seeing it because your script does not even advance to that point.

Now regarding the bigger issue. Since not even the LIDAR is spawning correctly, the reason could be that you did not load the sourcing script. Are you executing the following?
```
cd gym-gazebo/gym_gazebo/envs/installation/
bash turtlebot_setup.bash

```

If this is the issue, I will update the instructions for Ubuntu16 since I see that instruction is missing there. Please check instructions from [Ubunt18](https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#ubuntu-1804) and provide feedback if you think something else is missing.

cavinluo, 2019-01-30:
Dear @nzlz, thank you very much for your quick reply.
[image]

I tried, but the bot still did not move. Any other possibilities may cause the problem?
Thank you very much.

nzlz, 2019-01-30:
Make sure you are not running other gazebo or gzserver instances, `ps aux | grep gazebo` . If you have them, kill them and try again.

Please share the full trace of the libprotobuf error.

Which version of gazebo are you using?

cavinluo, 2019-01-30:
This time the error of libprotobuf did not show. 
Gazebo multi-robot simulator, version 7.0.0. Only one Gazebo running.

The full trace of the running is as following:
cai@cai-Lenovo-S40-70:~/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  result = entry_point.load(False)
ROS_MASTER_URI=http://localhost:12329

GAZEBO_MASTER_URI=http://localhost:12330

Gazebo launched!
... logging to /home/cai/.ros/log/1adaf9ba-2474-11e9-8694-acb57d8ed621/roslaunch-cai-Lenovo-S40-70-3093.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Unable to register with master node [http://localhost:12329]: master may not be running yet. Will keep trying.
Done checking log file disk usage. Usage is <1GB.

xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
xacro.py is deprecated; please use xacro instead
started roslaunch server http://cai-Lenovo-S40-70:43477/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /opt/ros/kinetic/...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [3109]
ROS_MASTER_URI=http://cai-Lenovo-S40-70:12329/

setting /run_id to 1adaf9ba-2474-11e9-8694-acb57d8ed621
process[rosout-1]: started with pid [3122]
started core service [/rosout]
process[gazebo-2]: started with pid [3125]
process[spawn_turtlebot_model-3]: started with pid [3135]
process[mobile_base_nodelet_manager-4]: started with pid [3139]
process[cmd_vel_mux-5]: started with pid [3144]
process[bumper2pointcloud-6]: started with pid [3145]
process[robot_state_publisher-7]: started with pid [3146]
/usr/lib/python2.7/dist-packages/matplotlib/backend_managers.py:62: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1 and perhaps the rcParam as well
  'version 2.1 and perhaps the rcParam as well')
/usr/lib/python2.7/dist-packages/matplotlib/backend_tools.py:78: UserWarning: Treat the new Tool classes introduced in v1.5 as experimental for now, the API will likely change in version 2.1, and some tools might change name
  'version 2.1, and some tools might change name')
[ INFO] [1548841675.867324317]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1548841675.867927926]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1548841678.676279286, 0.860000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1548841678.768054602, 47.230000000]: Physics dynamic reconfigure ready.
[spawn_turtlebot_model-3] process has finished cleanly
log file: /home/cai/.ros/log/1adaf9ba-2474-11e9-8694-acb57d8ed621/spawn_turtlebot_model-3*.log
[ INFO] [1548841679.557447431, 38.610000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1548841679.617328835, 38.610000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1548841679.792166336, 38.610000000]: Starting plugin Kobuki(ns = //)
[ WARN] [1548841679.792239214, 38.610000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1548841679.808119616, 38.610000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1548841679.808473444, 38.610000000]: Will publish tf. [mobile_base]
[ INFO] [1548841679.811016390, 38.610000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1548841679.812222618, 38.610000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1548841679.817429116, 38.610000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1548841679.826564838, 38.610000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1548841679.832657370, 38.610000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/velocity!
[ INFO] [1548841679.834772156, 38.610000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1548841679.836334965, 38.610000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1548841679.837914447, 38.610000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1548841679.837979312, 38.610000000]: GazeboRosKobuki plugin ready to go! [mobile_base]

[image]

nzlz, 2019-01-30:
Gazebo 7 is quite an old version, many things changed since the 8, I suggest you update. 

In order to discard issues with MASTER_URIs, please delete these two lines:

https://github.com/erlerobot/gym-gazebo/blob/cf30f243dcc177de246d5f9debbabd2665c9b098/gym_gazebo/envs/gazebo_env.py#L27-L28

cavinluo, 2019-01-31:
Dear @nzlz , after a long time struggling to solve the problem in 16.04, I upgrade the Ubuntu to 18.04.
Now I can see the bot  inside the maze but seems it still cannot work properly.
The errors show as following:
raise error.Error("Tried to reset environment which is not done. While the monitor is active for {}, you cannot call reset() unless the episode is over.".format(self.env_id))
gym.error.Error: Tried to reset environment which is not done. While the monitor is active for GazeboCircuit2TurtlebotLidar-v0, you cannot call reset() unless the episode is over.

Is there any solution for this problem?Thank you very much.
[image]

cavinluo, 2019-01-31:
Dear @nzlz , Thank you very very much!
The bot is happily searching the maze now.
Sorry that I have to skip the 16.04 version, reinstall and upgrade seemed the faster solution for me.

nzlz, 2019-01-31:
I'm glad its working fine now :) Make sure to open a new issue if you face any other problem. Closing.

closed 2019-01-31 by nzlz