erlerobot/gym-gazebo #207: Installation error during "bash turtlebot_setup.bash"

bsaitejaswi, 2019-02-03:
Hello,

I have launched turtlebot_setup.bash to load the environment variables for the turtlebot. I get the following error:
[image]

May I know how to solve this or where I am going wrong?

Also when I run a sample example for the turtlebot, I get the following error. I believe it is due to not loading the environment properly.
[image]
[image]

Any help will be greatly appreciated. Thanks!

nzlz, 2019-02-03:
Seems like you catkin_ws installation did not finish properly. Can you make sure you have the `/turtlebot_gazebo/env-hooks25.turtlebot-gazebo.sh.em` file? If you don't have it try getting the corresponding package and compile it.

---
If you keep getting problems with turtlebot_setup.bash ...

If you open turtlebot setup script you will find simple exports + couple of tweaks related to turtlebot description files.

Please open [turtlebot_setup.bash](https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/installation/turtlebot_setup.bash) with a text editor an try to replicate those command one by one.

Example: If you find a line like this..
```sh
if [ -z "$GAZEBO_MODEL_PATH" ]; then
  bash -c 'echo "export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:"`pwd`/../assets/models >> ~/.bashrc'
else
  bash -c 'sed "s,GAZEBO_MODEL_PATH=[^;]*,'GAZEBO_MODEL_PATH=`pwd`/../assets/models'," -i ~/.bashrc'
fi
```
You can do the following:
```sh
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:`pwd`/../assets/models >> ~/.bashrc
```

bsaitejaswi, 2019-02-03:
Hi,

> Seems like you catkin_ws installation did not finish properly. Can you make sure you have the `/turtlebot_gazebo/env-hooks25.turtlebot-gazebo.sh.em` file? If you don't have it try getting the corresponding package and compile it.

May i ask where should the file actually be located? Ia this is a suitable package to use: https://github.com/turtlebot/turtlebot

As for replication of commands I need to do it for the all the if statements?

Thanks!

bsaitejaswi, 2019-02-03:
Okay, I have all installed all the necessary packages in the gazebo.repos file inside gym_gazebo/envs/installation/catkin_ws/src. Now when when run the "bash turtlebot_setup.bash it works fine. Thanks!

However, just wondering now when I type "bash setup_kinetic.bash", I get this error. 
[image]
I have tried to catkin_make before this. 

This has also resulted in the similar error traceback when i try to run the example as shown:
[image]
[image]

nzlz, 2019-02-03:
seems like you did not install vcs, python3-vcstool. Did you skip those steps in the installation process? I sugest you follow the instructions using vcs, since it seems like you installed a wrong version of a repo. Or you can also try doing a checkout to the kinetic branch inside that repository.

bsaitejaswi, 2019-02-03:
Oh yes, I think I have missed it unknowingly. I have installed them now and for some such as the kobuki-core and turtlebot, I have changed it to checkout to the kinetic branch inside that repository.
So, those errors have been fixed now. 

Sorry to keep on bothering but now there has been other errors saying there are multiple packages with the same name when i do catkin_make and bash setup_kinetic.bash. Can I delete one of the duplicates?
[image]

nzlz, 2019-02-03:
I sugges you remove everything except the /src folder inside catkin_ws and compile again.

nzlz, 2019-02-03:
Remove the src folder too and follow from vcs import step again. Dont even try using the turtlebot_setup script if you did not compile the workspace correctly.

bsaitejaswi, 2019-02-03:
Yes I have tried and once again I get a similar error saying that there are 
"Multiple packages found with the same name "ecl_mobile_robot":
- ecl_core/ecl_mobile_robot
- ecl_lite/ecl_mobile_robot "

nzlz, 2019-02-04:
**ecl_mobile_robot** is only located in **ecl_core**. [ecl_lite](https://github.com/stonier/ecl_lite) does not contain such package. Please make sure you are in devel branch in both packages. I don't know how is it possible that this happened, I don't see `ecl_mobile_robot` even in other branches of `ecl_lite`... strange.

nzlz, 2019-02-04:
Can you remove `ecl_navigation` from catkin_ws/src and try again? I gues you will need to remove it form /build and /devel too.

bsaitejaswi, 2019-02-04:
Hmm okay I have removed the ecl_navigation and tried to do compile again but was still getting the same error.

So, I removed the catkin_ws, commented the ecl_navigation package in gazebo.repos file and then compiled again. Now there is a CMake error due to which the compilation is not possible.
[image]

nzlz, 2019-02-04:
Should work, but can't confirm that. Note that we are putting our efforts in releasing a new version, so this one is only community maintained now. I _guarantee_ ubuntu18 installation works fine, ubuntu16 _should_ work fine too.

bsaitejaswi, 2019-02-04:
Hi, I have actually managed to execute some commands as follows while I was searching more online about this:

#added these into the gazebo.repos file and commented the ecl_navigation
geometry:
   type: git
   url: https://github.com/ros/geometry
   version: indigo-devel
 geometry2:
   type: git
   url: https://github.com/ros/geometry2
   version: indigo-devel
 vision_opencv:
   type: git
   url: https://github.com/ros-perception/vision_opencv.git
   version: kinetic

#then tried to complie again, it failed
bash setup_kinetic.bash

#executed these commands below as found online
cd ~
touch gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/joystick_drivers/spacenav_node/CATKIN_IGNORE
touch gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/joystick_drivers/wiimote/CATKIN_IGNORE
touch gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/kobuki_desktop/kobuki_qtestsuite/CATKIN_IGNORE

#now when i tried to compile again it is successful i presume or maybe not? screenshots of the terminal
[image]

nzlz, 2019-02-04:
You cant execute the automatic installation script if you are using custom fixed. Open the setup kinetic file with a text editor and try to follow the instructions to compile it. 

Also, mav_msgs should be there, but i believe you cam skip it if you are going to use the Turtlebot.

bsaitejaswi, 2019-02-07:
Hi, 

I have tried the fix the error by following the fix in issue #110 and now it does not give the error anymore. I have also removed the ecl_navigation package to avoid the multiple packages with same name error.

However, now when I type bash setup_kinetic.bash, I get the following error:
[image]

How should I go about fixing it?

nzlz, 2019-02-07:
clone this to your directory:

https://github.com/ros-perception/ar_track_alvar/tree/kinetic-devel (kinetic-devel branch)

bsaitejaswi, 2019-02-07:
I have cloned the above repository into the following path gym-gazebo/envs/installation/catkin_ws/src and tried to catkin_make and bash setup_kinetic bash again, now I get a really strange error.
[image]

nzlz, 2019-02-08:
Not having `catkin_ws/devel/setup.bash` means you did not compile the carkin workspace (catkin_make). Or that you placed your workspace in a different folder, in that case update the reference to catkin_ws/devel/setup.bash in yout **~/.bashrc**. `turtlebot_setup.bash` file is not the issue, it just calls bash at the end which is failing for you.

bsaitejaswi, 2019-02-08:
Hi, I have figured the error, it was a line in the ~/.bashrc file that was causing the error. I have recompiled again and now the bash turtlebot_setup.bash works fine. Now when I try to run the example, it gives a new error that no module named gym_gazebo! I have followed the installation steps and the output is the same. Should I open a new issue for this?

nzlz, 2019-02-08:
Please continue this issue. You have to install gym-gazebo: https://github.com/erlerobot/gym-gazebo/blob/master/INSTALL.md#gazebo-gym

bsaitejaswi, 2019-02-08:
Yes I have installed it following the instructions and this is the output i get:
[image]

I still get the same error when I try to run it.

bsaitejaswi, 2019-02-10:
Hi sorry to bother you again but I tried to install gym-gazebo in the root repository and run the example once again and I get the following error.
[image]

nzlz, 2019-02-10:
Did you execute `source turtlebot_setup.bash` before launching? This will add a custom urdf, check last lines of that script.

bsaitejaswi, 2019-02-10:
Yes I did and it did not give any error. Yes, the bash does have the urdf file, I did not change anything from the original file though.

These are the last few lines:

#copy altered urdf model
cp -r ../assets/urdf/kobuki_urdf/urdf/ catkin_ws/src/kobuki/kobuki_description

#copy laser mesh file
cp ../assets/meshes/lidar_lite_v2_withRay.dae catkin_ws/src/kobuki/kobuki_description/meshes
 
exec bash # reload bash

nzlz, 2019-02-10:
https://github.com/erlerobot/gym-gazebo/issues/58

Take a look at that issue, i believe that will solve your problem.

bsaitejaswi, 2019-02-10:
Yes that solved the error but the i still cannot launch the example properly. I get a new error now and it remains the same for the long time.

tejaswi@tejaswi-Inspiron-3559:~/gym-gazebo/examples/turtlebot$ python circuit2_turtlebot_lidar_qlearn.py 
ROS_MASTER_URI=http://localhost:11592

GAZEBO_MASTER_URI=http://localhost:11593

Gazebo launched!
Unable to register with master node [http://localhost:11592]: master may not be running yet. Will keep trying.
... logging to /home/tejaswi/.ros/log/65332016-2d73-11e9-bd34-2c6e857b2ca8/roslaunch-tejaswi-Inspiron-3559-17646.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://tejaswi-Inspiron-3559:46330/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/tejaswi/gym...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [17668]
ROS_MASTER_URI=http://tejaswi-Inspiron-3559:11592/

setting /run_id to 65332016-2d73-11e9-bd34-2c6e857b2ca8
Traceback (most recent call last):
  File "circuit2_turtlebot_lidar_qlearn.py", line 25, in <module>
    env = gym.make('GazeboCircuit2TurtlebotLidar-v0')
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 171, in make
    return registry.make(id, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 123, in make
    env = spec.make(**kwargs)
  File "/usr/local/lib/python2.7/dist-packages/gym/envs/registration.py", line 87, in make
    env = cls(**_kwargs)
  File "/home/tejaswi/gym-gazebo/gym_gazebo/envs/turtlebot/gazebo_circuit2_turtlebot_lidar.py", line 26, in __init__
    self.action_space = spaces.Discrete(3) #F,L,R
  File "/usr/local/lib/python2.7/dist-packages/gym/spaces/discrete.py", line 15, in __init__
    super().__init__((), np.int64)
TypeError: super() takes at least 1 argument (0 given)
tejaswi@tejaswi-Inspiron-3559:~/gym-gazebo/examples/turtlebot$ process[rosout-1]: started with pid [17696]
started core service [/rosout]
process[gazebo-2]: started with pid [17705]
process[spawn_turtlebot_model-3]: started with pid [17724]
process[mobile_base_nodelet_manager-4]: started with pid [17725]
process[cmd_vel_mux-5]: started with pid [17726]
process[bumper2pointcloud-6]: started with pid [17727]
process[robot_state_publisher-7]: started with pid [17743]
[ INFO] [1549830929.483838442]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1549830929.487457175]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1549830930.631382791, 0.590000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1549830931.562306965, 0.720000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1549830931.571335899, 0.720000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1549830931.713996146, 0.720000000]: Physics dynamic reconfigure ready.
[ INFO] [1549830931.783753518, 0.720000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1549830931.784163826, 0.720000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1549830931.821894175, 0.720000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1549830931.836369866, 0.720000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1549830931.838154637, 0.720000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1549830931.842146909, 0.720000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1549830931.850884442, 0.720000000]: Will publish tf. [mobile_base]
[ INFO] [1549830931.859398513, 0.720000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1549830931.864767236, 0.720000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1549830931.895265217, 0.720000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1549830931.919913545, 0.720000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1549830931.929457073, 0.720000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1549830931.932057415, 0.720000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1549830931.937941310, 0.720000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1549830931.942376238, 0.720000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1549830931.942590359, 0.720000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
[spawn_turtlebot_model-3] process has finished cleanly
log file: /home/tejaswi/.ros/log/65332016-2d73-11e9-bd34-2c6e857b2ca8/spawn_turtlebot_model-3*.log

nzlz, 2019-02-11:
Launch the GUI so that you can see the problem.

- Check this part of the README : https://github.com/erlerobot/gym-gazebo#display-the-simulation
- You can choose that option to spawn the GUI, or set GUI to True in this line : https://github.com/erlerobot/gym-gazebo/blob/176b61c6e576d5dce7ab1ceba6d0226c9beda847/gym_gazebo/envs/assets/launch/GazeboCircuit2TurtlebotLidar_v0.launch#L6

Regarding the issue, I would say the `step` function in your environment is not being executed properly. Could be because you did not source `turtlebot_setup.bash` and the LIDAR is not taking correct readings.

bsaitejaswi, 2019-02-11:
Hi, thanks! Now the gazebo launches with the environment and turtlebot as well. However, the turtlebot does not move and the simulation does not start. Yes, I did source the turtlebot_setup.bash

nzlz, 2019-02-11:
Oh I missed your previous error sorry. You have a registration issue with the Gym.

Since you installed gym_gazebo correctly, I believe this could be a gym issue. Please uninstall Gym, if you installed via pip, you can `pip uninstall gym`. Then install Gym from sources:

```
cd ~
git clone https://github.com/openai/gym
cd gym
pip3 install -e .
```

**Note**: The installation process if taking much much longer than expected in your case. Of course you can make it work in Ubuntu16, but once you manage to do it, please document what you do if you intend to work in this platform. I believe going with Ubuntu18 would be an easier path, consider it for the future.

chenjiunhan, 2019-02-21:
> Okay I reinstallled the gym-gazebo again and now when I try to bash setup_kinetic.bash, it still says there are multiple packages with the same name.
> Multiple packages found with the same name "ecl_mobile_robot":
> 
> * ecl_core/ecl_mobile_robot
> * ecl_navigation/ecl_mobile_robot

According to the link below, the error message above is caused by the upgrade of ecl.
Change the version of ecl_core, ecl_lite and ecl_tools to release/0.61-indigo-kinetic in 
"gym-gazebo/gym_gazebo/envs/installation/gazebo.repos" can solve the problem.

Warning: the content is written in Mandarin.
https://blog.csdn.net/Hansry/article/details/81057945

cgr71ii, 2019-02-23:
> According to the link below, the error message above is caused by the upgrade of ecl.
> Change the version of ecl_core, ecl_lite and ecl_tools to release/0.61-indigo-kinetic in [...]

[Here](https://github.com/erlerobot/gym-gazebo/issues/110) is commented the same problem (see https://github.com/erlerobot/gym-gazebo/issues/110#issuecomment-384653231).

HareshMiriyala, 2019-03-01:
@chenjiunhan  That worked ! Thanks !

nzlz, 2019-03-01:
@HareshMiriyala a pull request would be welcomed with the changes you needed to make it work in Ubuntu16.

closed 2019-03-01 by nzlz

HareshMiriyala, 2019-03-01:
[image]

Getting this weird error when i run 'bash turtlebot_setup.bash' . I ran 'bash setup_kinect.bash' and it worked with no errors.

HareshMiriyala, 2019-03-01:
When i run the example  'python3 maze_turtlebot_lidar_qlearn.py ' , i get a grey plot with no points and some bunch of rewards printed in the console. But i don't see a gazebo simulation

nzlz, 2019-03-01:
To display the simulation: https://github.com/erlerobot/gym-gazebo#display-the-simulation

But first you must source correctly the turtlebot_setup.bash. Did you install catkin_ws inside the installation folder?

HareshMiriyala, 2019-03-01:
yes, i did