erlerobot/gym-gazebo #210: 4 legged mekamon example

brothergomez, 2019-02-11:
Hi all,

I am attempting to get gym-gazebo working, initially just with a test environment with a "Mekamon" robot model: [https://mekamon.com/ ](https://mekamon.com/ ). I have forked the gym-gazebo repo and the urdf.xacro file is ready made inside the gym_gazebo/envs/assets/urdf/mekamon folder. Here is the forked repo: [https://github.com/brothergomez/gym-gazebo](url)

So far, I have made an attempt to do the basic setup for the robot: 

1. Registering the environment in an init file
2. Made launch files, and put urdf files in the correct places. I can build the catkin workspace with no errors. 
3. Creating an environment file, loosely based off the turtlebot and ARIAC examples
4. Create an Adam example using the above environment, loosely based off the cartpole example.

However, when running the example I get no errors, but the model spawns in and doesn't move. I would at least expect some movement. Have I correctly linked everything up? Am I using the correct publisher and subscribers, I really don't understand these?

I can run the turtlebot lidar example with no issues. 

**Robot**
Mekamon - details available in forked repository:
[https://github.com/brothergomez/gym-gazebo](url)

**Target**
Getting a gym environment working with a 4 legged robot. Reward based on distance travelled from the initial position. Rewards etc ideally based on gym "Ant" example: [Ant](https://github.com/openai/gym/blob/master/gym/envs/mujoco/ant.py)

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04
 - ROS version: ROS Kinetic
 - Gazebo version: 7.14.0

**About you**
Name the organization your represent, if proceeds.

nzlz, 2019-02-11:
Hi @brothergomez , first of all thanks for using gym-gazebo!

I will need some extra information regarding the exact point where you are stuck. I believe you are stuck in the **step** function, if not, please tell me the exact point with a pointer to the code. Inside the step function is where you:

1. select an action
2. execute that action (publish ros message)
3. Observe the environment
4. Compute reward + return state, reward etc

Once you unpause physics (maybe you don't even need to do this, but definitely won't hurt), you have to select + execute the action. I see you have already defined **self._pub**, so you have to use it at this point. Now, depending on the algorithm you are going to run, you can:
1. Wait to the action to be completed. This is your choice on how to implement it. I would go with this one.
2. Don't wait, just keep with the flow. Could be positive for a faster convergence is some specific algorithms.

If you chose n1, once your action is executed, you are ready to observe the environment. I can see you already have that observation in a while loop. So after you observe, you need to compute the reward and tune it for best behavior.

brothergomez, 2019-02-12:
Hi @nzlz , many thanks for your help.

I think my issue is with how to publish a ros message as I'm not sure what the correct syntax is for publishing. 

I've tried `self._pub.publish(self.get_trajectory_message(action[:12]))` , to no avail. I'm a bit confused about what is passed to the step function in the action variable? 

On the turtlebot I see it has a `discretize_observation` function - I'm guessing this turns the continuous observation data into discrete data for easier processing? In my case I guess I don't need this - is my observation data just the angles of the joints? 

I've currently got my observation and action spaces both set at a 12 dimension box - 1 for each joint angle. However my Jointangles msg is the following:

`uint16 front_left_knee`<br/>
`uint16 front_left_thigh `<br/>
`uint16 front_left_hip`<br/>
`uint16 front_right_knee`<br/>
`uint16 front_right_thigh`<br/>
`uint16 front_right_hip`<br/>
`uint16 back_left_knee`<br/>
`uint16 back_left_thigh`<br/>
`uint16 back_left_hip`<br/>
`uint16 back_right_knee`<br/>
`uint16 back_right_thigh`<br/>
`uint16 back_right_hip`<br/>

Does this mean I should have a discrete action/observation space? 

I have added the logfiles to a gist here if it helps: [https://gist.github.com/brothergomez/c16dd589f48182940b036682736a47f3](gist)

nzlz, 2019-02-12:
I'm afraid I can't take a deep look at this currently. But regarding the observation, since you want to train to robot to walk, you will need the parameters you mention indeed.

The available MARA environments are also aware of the pose of the joints, meaning you can find useful code for your own application there. Specially:

https://github.com/erlerobot/gym-gazebo/blob/9bdfe1ec8c377216601680b7cadf9bf73af79598/gym_gazebo/envs/MARA/gazebo_mara_top_orient_collision_v0.py#L263-L265