erlerobot/gym-gazebo #211: spawn_model fails running circuit_turtlebot_lidar_qlearn.py

MrDadaGuy, 2019-02-18:
I'm on Ubuntu 18.04, ROS Melodic.  Python 2.7.15rc1

I'm trying to run the sample from INSTALL.md, when I run python circuit_turtlebot_lidar_qlearn.py, I get an error:

```
[ERROR] [1550501519.367288, 38.790000]: Spawn service failed. Exiting.
[spawn_turtlebot_model-3] process has died [pid 27417, exit code 1, cmd /home/modsim/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -unpause -urdf -param robot_description -model mobile_base __name:=spawn_turtlebot_model __log:=/home/modsim/.ros/log/bc7ff19a-338c-11e9-8e49-705a0f42eb8d/spawn_turtlebot_model-3.log].
log file: /home/modsim/.ros/log/bc7ff19a-338c-11e9-8e49-705a0f42eb8d/spawn_turtlebot_model-3*.log
```

I am able to run gazebo standalone, and load lots of models from the model warehouse.

Thanks!

MrDadaGuy, 2019-02-18:
Adding some additional info.  I found it interesting that it's spinning up a new ROS MASTER...

modsim@modsimserver:~/catkin_ws/src/gym-gazebo/examples/turtlebot$ python circuit_turtlebot_lidar_qlearn.py
ROS_MASTER_URI=http://localhost:13266

GAZEBO_MASTER_URI=http://localhost:13267

Gazebo launched!
Unable to register with master node [http://localhost:13266]: master may not be running yet. Will keep trying.
... logging to /home/modsim/.ros/log/bc7ff19a-338c-11e9-8e49-705a0f42eb8d/roslaunch-modsimserver-27370.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://modsimserver:42437/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/modsim/catk...
 * /gazebo/enable_ros_network: True
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [27385]
ROS_MASTER_URI=http://modsimserver:13266/

setting /run_id to bc7ff19a-338c-11e9-8e49-705a0f42eb8d
process[rosout-1]: started with pid [27403]
started core service [/rosout]
process[gazebo-2]: started with pid [27410]
process[spawn_turtlebot_model-3]: started with pid [27417]
process[mobile_base_nodelet_manager-4]: started with pid [27418]
process[cmd_vel_mux-5]: started with pid [27421]
process[bumper2pointcloud-6]: started with pid [27422]
process[robot_state_publisher-7]: started with pid [27427]
[ INFO] [1550501518.535696327]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1550501518.537622957]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1550501519.130451506, 0.950000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1550501519.160086358, 10.480000000]: Physics dynamic reconfigure ready.
[ERROR] [1550501519.367288, 38.790000]: Spawn service failed. Exiting.
[spawn_turtlebot_model-3] process has died [pid 27417, exit code 1, cmd /home/modsim/catkin_ws/src/gazebo_ros_pkgs/gazebo_ros/scripts/spawn_model -unpause -urdf -param robot_description -model mobile_base __name:=spawn_turtlebot_model __log:=/home/modsim/.ros/log/bc7ff19a-338c-11e9-8e49-705a0f42eb8d/spawn_turtlebot_model-3.log].
log file: /home/modsim/.ros/log/bc7ff19a-338c-11e9-8e49-705a0f42eb8d/spawn_turtlebot_model-3*.log
[ INFO] [1550501519.788303963, 92.430000000]: Camera Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1550501519.794781519, 92.430000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1550501519.862286621, 92.430000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1550501519.862392451, 92.430000000]: Starting Laser Plugin (ns = /)
[ INFO] [1550501519.864537138, 92.430000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1550501519.973255676, 92.430000000]: Starting plugin Kobuki(ns = //)
[ WARN] [1550501519.973320058, 92.430000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1550501519.974288498, 92.430000000]: Kobuki(ns = //): <tf_prefix> = 
[ INFO] [1550501519.974450602, 92.430000000]: Will publish tf. [mobile_base]
[ INFO] [1550501519.976208556, 92.430000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1550501519.976965678, 92.430000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1550501519.981736620, 92.430000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1550501519.984799265, 92.430000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1550501519.987065095, 92.430000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/velocity!
[ INFO] [1550501519.987578189, 92.430000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1550501519.988063626, 92.430000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1550501519.988563053, 92.430000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1550501519.988591893, 92.430000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
EP: 1 - [alpha: 0.1 - gamma: 0.8 - epsilon: 0.9] - Reward: -157     Time: 0:00:02
EP: 2 - [alpha: 0.1 - gamma: 0.8 - epsilon: 0.9] - Reward: 28     Time: 0:00:05
EP: 3 - [alpha: 0.1 - gamma: 0.8 - epsilon: 0.9] - Reward: -48     Time: 0:00:07
EP: 4 - [alpha: 0.1 - gamma: 0.8 - epsilon: 0.9] - Reward: -45     Time: 0:00:10
EP: 5 - [alpha: 0.1 - gamma: 0.8 - epsilon: 0.9] - Reward: -157     Time: 0:00:11

nzlz, 2019-02-18:
Hi @MrDadaGuy, is the issue solved?

MrDadaGuy, 2019-02-18:
No, it's not.. I still have this problem.  I just posed additional console output in the hopes it might be helpful.

nzlz, 2019-02-18:
Oh okay, skipped the error. Did you source turtlebot_setup.bash? 

Can you launch the GUI and check whats happening visually? Intructions are available in the readme.

MrDadaGuy, 2019-02-18:
Yes I am sourcing turtlebot_setup.bash.  Oddly, I do the same in a different terminal window, and rosparam list says, Unable to communicate with master!  

I'll try to find the instructions on how to launch the GUi.  Thanks

MrDadaGuy, 2019-02-18:
Some progress --  same error shows up.  However, I can now get the GUI running!  
I export the GAZEBO_MASTER_URI and the *second* ROS_MASTER_URI (the one that shows up right above the SUMMARY) and now I have the model showing up.  
But one step backward -- now the program stops on [ INFO] [1550507402.958690828, 33.020000000]: GazeboRosKobuki plugin ready to go! [mobile_base] .. it used to bring up matplotlib window and that's no longer happening...

nzlz, 2019-02-18:
Can you `killall gzserver `and try again?

MrDadaGuy, 2019-02-18:
Does not help, this is repeatable.

MrDadaGuy, 2019-02-18:
Perhaps I should say that this issue is solved now, since I am able to get Gazebo running with the model.  I'll start a new issue describing how the learning process is not starting.  

In summary, my issue was the spawn_model ERROR showing up on console when I run the python program.  I think what I've learned is that this is inconsequential, but to get gazebo running I need to (1) export ROS_MASTER_URI, and (2) export GAZEBO_MASTER_URI, and (3) run gzclient.

nzlz, 2019-02-19:
In the last part of this small section, you can see that you can remove the environment variables manually to avoid this process. If you don't need the automatic network segmentation. Remove [these](https://github.com/erlerobot/gym-gazebo/blob/9bdfe1ec8c377216601680b7cadf9bf73af79598/gym_gazebo/envs/gazebo_env.py#L27-L28) 2 lines.

About your error, I notice your `catkin_ws` is not in the place it is supposed to be, inside _envs/installation_. Anyway, if you don't want to go through the installation process again, you can modify [these](https://github.com/erlerobot/gym-gazebo/blob/9bdfe1ec8c377216601680b7cadf9bf73af79598/gym_gazebo/envs/installation/turtlebot_setup.bash#L10-L11) two lines to include the absolute path to your location of _catkin_ws_.

Finally, make sure you launch the script from script folder, not other one.

closed 2019-03-12 by nzlz