erlerobot/gym-gazebo #212: Run gym-gazebo on Google Colaboratory

talregev, 2019-02-21:
**Robot**
Name of the robot. Provide a reference if the robot is not officially supported by gym-gazebo.

**Feature description**
Provide a detailed description about the feature you would like to have implemented in gym-gazebo.   

I am trying to run gym-gazebo on Google Colaboratory.     
There is a problem to run gazebo server (gazebo without gui) on Colab.    
There was warning on display: `Unable to create X window. Rendering will be disabled`  but I think I fix it.   
There also warning about sound card. not sure it related `Unable to open audio device`,     
And there warning about conversion `Conversion of sensor type[depth] not supported`    
Here is an [example][1]     
Can you help?

[1]:https://colab.research.google.com/drive/1Q8P-i8Ih3qkQUv572hv3rgf8EhRRUKjn

nzlz, 2019-02-21:
Hi @talregev , interesting stuff! 

I don't have any experience with Colab myself, but I can try to help you.

I don't know why something is trying to launch the ALSA audio drivers, but seems like it does not interfere with the execution of gym-gazebo.

Since you are going with a more advanced project, Ill suppose you already made the Turtlebot work with your current changes in a standard setup.
If you are running with no GUI, the plot launched (matplotlib) could be causing some problems. Also I cant find where you are sourcing the Turtlebot provisioning script. Please take a deep look at the `turtlebot_setup.bash` script, specially [these](https://github.com/erlerobot/gym-gazebo/blob/9bdfe1ec8c377216601680b7cadf9bf73af79598/gym_gazebo/envs/installation/turtlebot_setup.bash#L10-L11) lines.

talregev, 2019-02-23:
I Succeed to run 3 example of turtlebot  on google colab:
[circuit_turtlebot_lidar_qlearn.py][1]
[circuit2_turtlebot_lidar_dqn.py][2]
[circuit2c_turtlebot_camera_dqn.py][3]

> Please take a deep look at the `turtlebot_setup.bash` script, specially [these](https://github.com/erlerobot/gym-gazebo/blob/9bdfe1ec8c377216601680b7cadf9bf73af79598/gym_gazebo/envs/installation/turtlebot_setup.bash#L10-L11) lines.

You are right, I didn't pay attention deep to this file. 

> I don't know why something is trying to launch the ALSA audio drivers, but seems like it does not interfere with the execution of gym-gazebo.

gzserver is open audio driver, but it not important to run the simulation even if it fail to load it. 

I want suggest suggestions and bug fix in order to run these example properly:   

1. bugfix: in [deepq.py line 9][4] change from `from examples.turtlebot import memory` to `import memory`
2. suggestion: sometimes you want to disable the random port allocation and run your own port:
in [gazebo_env.py line 32-33][5] change to       
 ```python
        print("ROS_MASTER_URI="    + os.environ["ROS_MASTER_URI"] + "\n")
        print("GAZEBO_MASTER_URI=" + os.environ["GAZEBO_MASTER_URI"] + "\n")
```
3. bug fix: in [circuit2c_turtlebot_camera_dqn.py line 19][6] change to `from keras.initializers import normal` in [circuit2c_turtlebot_camera_dqn.py line 30][7] add 
```python
from keras import backend as K
K.set_image_dim_ordering('th')
```
4. suggestion: In these example files [circuit_turtlebot_lidar_qlearn.py][1] [circuit2_turtlebot_lidar_dqn.py][2] [circuit2c_turtlebot_camera_dqn.py][3] add output dir parameter.

5. suggestion: In these files: [GazeboCircuitTurtlebotLidar_v0.launch][8] [GazeboCircuit2TurtlebotLidar_v0.launch][9] [GazeboCircuit2cTurtlebotLidar_v0.launch][10] add line `<arg name="verbose" value="true"/>` for more information gazebo (gzserver) is output.

6. bugfix: In these files [lidar_sensor.urdf.xacro line 59][11] [lidar_sensor.urdf.xacro line 59][12] [lidar_sensor.urdf.xacro line 59][13] change `librotors_gazebo_sonar_plugin.so` to `libhector_gazebo_ros_sonar.so`. There is not rotors_gazebo_sonar_plugin only hector_gazebo_ros_sonar plugin. 

[1]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/circuit_turtlebot_lidar_qlearn.py
[2]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/circuit2_turtlebot_lidar_dqn.py
[3]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/circuit2c_turtlebot_camera_dqn.py
[4]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/deepq.py#L9
[5]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/gazebo_env.py#L32-L33
[6]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/circuit2c_turtlebot_camera_dqn.py#L19
[7]:https://github.com/erlerobot/gym-gazebo/blob/master/examples/turtlebot/circuit2c_turtlebot_camera_dqn.py#L30
[8]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/launch/GazeboCircuitTurtlebotLidar_v0.launch#L15
[9]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/launch/GazeboCircuit2TurtlebotLidar_v0.launch#L15
[10]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/launch/GazeboCircuit2cTurtlebotLidar_v0.launch#L16
[11]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/urdf/kobuki_cnn_urdf/kobuki_urdf/urdf/sensors/lidar_sensor.urdf.xacro#L59
[12]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/urdf/kobuki_nn_urdf/urdf/sensors/lidar_sensor.urdf.xacro#L59
[13]:https://github.com/erlerobot/gym-gazebo/blob/master/gym_gazebo/envs/assets/urdf/kobuki_urdf/urdf/sensors/lidar_sensor.urdf.xacro#L59

talregev, 2019-02-23:
I want to able in gazebo (gzserver) record every epoch a video, that I will able to see what the robot is doing every epoch on google colab. 

Can you help?

nzlz, 2019-02-23:
Can you make a PR with the changes you listed? they all seem good to me except n2. That one we will leave as is for now.

Gazebo has an option to record complete log file, which later you can replay. If you type `gazebo --help` you will find it. I have never used it but should work for you, let me know if you need more help with this.

talregev, 2019-02-23:
I make PR.

> Gazebo has an option to record complete log file, which later you can replay. If you type `gazebo --help` you will find it. I have never used it but should work for you, let me know if you need more help with this.

thinking of create inside camera above that send images of the world.

talregev, 2019-03-12:
i succeeded to run inside google colab. so i am close this issue.

closed 2019-03-12 by talregev