erlerobot/gym-gazebo #213: Installation error during "bash setup_kinetic.bash"

chenjiunhan, 2019-02-21:
**Describe the bug**
The bug is NOT from this repository, just for reference. 
When I executed the following command, vcs python3-vcstool crashed:
bash setup_kinetic.bash

I thought I missed some steps, but it seems that I should upgrade my setuptools as the discussion in this link:
https://github.com/pypa/pip/issues/4935

I executed pip3 install --user --upgrade setuptools and reinstall python3-vcstool, and then the problem was solved.

**Error trace**
error of vcs command in setup_kinetic.bash:

Traceback (most recent call last):
  File "/usr/bin/vcs", line 6, in <module>
    from pkg_resources import load_entry_point
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2927, in <module>
    @_call_aside
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2913, in _call_aside
    f(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2952, in _initialize_master_working_set
    add_activation_listener(lambda dist: dist.activate())
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 956, in subscribe
    callback(dist)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2952, in <lambda>
    add_activation_listener(lambda dist: dist.activate())
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2515, in activate
    declare_namespace(pkg)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2097, in declare_namespace
    _handle_ns(packageName, path_item)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2047, in _handle_ns
    _rebuild_mod_path(path, packageName, module)
  File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2066, in _rebuild_mod_path
    orig_path.sort(key=position_in_sys_path)
AttributeError: '_NamespacePath' object has no attribute 'sort'

**Desktop (please complete the following information):**
 - OS: Ubuntu 16.04

nzlz, 2019-02-21:
Thanks for the report @chenjiunhan ,
Is this report just informative or are you still facing issues?

chenjiunhan, 2019-02-21:
It's just informative, you can close this issue.

nzlz, 2019-02-21:
Alright thanks a lot! Its good to keep track of the issue in case it happens to someone else.

closed 2019-02-21 by nzlz