erlerobot/gym-gazebo #217: Update to Docker launch instructions

MrDadaGuy, 2019-02-27:
The instructions for connecting gazebo to docker did not work for me, so I did the following, hopefully this helps someone.

```
export GAZEBO_ID=`docker ps | grep gym-gazebo | awk '{print $1}'`
export GAZEBO_MASTER_IP=$(sudo docker inspect -f '{{range .NetworkSettings.Networks}}{{.IPAddress}}{{end}}' $GAZEBO_ID)
export GAZEBO_MASTER_URI=$GAZEBO_MASTER_IP:11345
```

LanderU, 2019-02-27:
Which are the outputs of those variables?

MrDadaGuy, 2019-02-27:
According to INSTALL document, I connect gzclient to docker instance using :  
```
export GAZEBO_MASTER_IP=$(sudo docker inspect --format '{{ .NetworkSettings.IPAddress }}' "id of running container")
export GAZEBO_MASTER_URI=$GAZEBO_MASTER_IP:11345

```
This did not work for me, and so I'm providing my commands above for the benefit of anyone else who has the same challenge.  I'm not asking a question otherwise, thanks!

commit ba2058fa (2019-02-28): Update INSTALL.md

Fix #217
closed 2019-02-28 by LanderU