erlerobot/gym-gazebo #218: Unable to use the turtlebot environment with stable baselines

HareshMiriyala, 2019-03-02:
**Robot**
GazeboCircuitTurtlebotLidar_v0

**Environment/Algorithm**
env = gym.make('GazeboCircuitTurtlebotLidar-v0')

**Describe the bug**
I declare the environment and use TRPO (that can support discrete action and discrete observation). Btw, i realized in the init function of the above environment, the self.observation_space = spaces.Discrete(5) was not declared and I have declared it. 

Now, stable baselines throws an error saying : 

ValueError: Cannot feed value of shape (1, 5) for Tensor 'input/Ob:0', which has shape '(?,)'

**Error trace**

`Traceback (most recent call last):
  File "/home/haresh/PycharmProjects/gazebo_simulations/main.py", line 18, in <module>
    model.learn(total_timesteps=25000)
  File "/usr/local/lib/python3.6/dist-packages/stable_baselines/trpo_mpi/trpo_mpi.py", line 318, in learn
    seg = seg_gen.__next__()
  File "/usr/local/lib/python3.6/dist-packages/stable_baselines/trpo_mpi/utils.py", line 57, in traj_segment_generator
    action, vpred, states, _ = policy.step(observation.reshape(-1, *observation.shape), states, done)
  File "/usr/local/lib/python3.6/dist-packages/stable_baselines/common/policies.py", line 452, in step
    {self.obs_ph: obs})
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/client/session.py", line 929, in run
    run_metadata_ptr)
  File "/usr/local/lib/python3.6/dist-packages/tensorflow/python/client/session.py", line 1128, in _run
    str(subfeed_t.get_shape())))
ValueError: Cannot feed value of shape (1, 5) for Tensor 'input/Ob:0', which has shape '(?,)'
`
**Screenshots**

**Desktop (please complete the following information):**
 - OS: ubuntu 16.04
 - ROS version Ros kinetic
 - Gazebo version  gazebo 8

HareshMiriyala, 2019-03-02:
It magically works when i instead define the observation space as continuous (Box of size 5, from -inf to inf)

nzlz, 2019-03-12:
Thanks for the feedback @HareshMiriyala , I mark this as solved.

closed 2019-03-12 by nzlz