erlerobot/gym-gazebo #219: How do i create my own custom environment ?

HareshMiriyala, 2019-03-08:
Built my own custom robot in gazebo, I have the urdf description file for the robot and the .world file for the world in which the robot needs to be placed. It would be great if there is some kind of writeup to know where to place the models and how to spawn our robot.

nzlz, 2019-03-09:
Hi @HareshMiriyala ,
We will be releasing a new version of the toolkit which uses ROS2 soon. So the guide you mention (which I agree would be interesting to make) would be for the new gym-gazebo. 

If you want to keep using the current ROS1 gym-gazebo, I recommend you take a look at the Turtlebot and try to understand how the process works with that robot. 

To your specific questions about urdf and world. Take a look at https://github.com/erlerobot/gym-gazebo/tree/master/gym_gazebo/envs/assets . Note that this structure is not convenient and will be changed completely.

HareshMiriyala, 2019-03-13:
Cool. I'll check the turtlebot env and will post questions here