erlerobot/gym-gazebo #222: ROS stops responding

Ignat-Georgiev, 2019-03-10:
Hi,

We're trying to run `gym_gazebo` to teach an autonomous racecar to drive around a race track.

The robot is a standard Ackermann car which is controlled with steering and velocity. The environment is a flat plane with traffic cones in the shape of a track. https://gitlab.com/eufs/eufs_sim

We're using Ubuntu 16.04 with ROS Kinetic and Gazebo 9.6. For the RL side, we're using the DDPG algorithm from [stable_baselines](https://stable-baselines.readthedocs.io/en/master/index.html).

We have created our own custom `gym_gazebo` environment and it seems to be working fine, however, ROS simply stops responding after some time and the project gets stuck waiting for ROS messages. The time until ROS becomes unresponsive varies greatly anytime between 10 min to 1h and is not very deterministic. No error messages are given and Gazebo keeps working as normal.

Has anybody encountered this issue before?

nzlz, 2019-03-11:
Hi @Ignat-Georgiev ,

One of the reasons for the switch to ROS2 in the upcoming release, is that ROS2 offers quality of service tuning for communications. I guess you are publishing/subscribing at high rates, so it could be that ROS simple gets too much information to handle at once.

Don't know any detail about your code, but I would suggest to try to make things slower and publish less frequently.

Ignat-Georgiev, 2019-03-14:
Thanks @nzlz 

ROS has a bit of a bad reputation with high-frequency data right?

In the original post, we were indeed publishing data at high frequencies. Odometry (position) data was being published at 200Hz, Camera images at 100Hz and the joint updates at 50Hz. It's worth noting that this is usually how we run this simulation for other purposes as well but haven't encountered such. Anyway, we still tried your suggestion of reducing the publication rates of these topics, but sadly that didn't fix the issue.

Completely by accident, we added an RViz node to visualise our progress and whenever that is running, the simulation never stops. (Or at least hasn't stopped in the last 36 hours of continuous running). I don't have an explanation as to why and how this solved the issue.

nzlz, 2019-03-14:
Regarding data frequency, i last time i checked we were faster than 1khz in our ROS2 tests in the upcoming version of gym-gazebo. This required some tune and being really careful with how data was handled,.. which parts were triggered when this data was received etc.

How did you reduce the publish rate? Are you using gazebos OnUpdate function?