erlerobot/gym-gazebo #224: Question: The robot move slowly and when I run it shows something wrong?

zhaolongkzz, 2019-03-10:
Thanks for sharing your code. Here I encountered some problem.
After I have run both the setup_kinetic and turtle_setup.bash, then the file circuit2_turtlebot_lidar_qlearn.py, I got a problem: `Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.`, and it influence that `[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported. [Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.`
So the robot move very slowly and not to detect the circle world.

Ubuntu16.04, ROS kinetic, python2.7

```bash
python circuit2_turtlebot_lidar_qlearn.py 
/home/zzl/RL/gym/gym/envs/registration.py:14: PkgResourcesDeprecationWarning: Parameters to load are deprecated.  Call .resolve and .require separately.
  result = entry_point.load(False)
ROS_MASTER_URI=http://localhost:12861

GAZEBO_MASTER_URI=http://localhost:12862

Gazebo launched!
Unable to register with master node [http://localhost:11311]: master may not be running yet. Will keep trying.
... logging to /home/zzl/.ros/log/7f5d39dc-4343-11e9-9b7f-0f45a3b6c431/roslaunch-zzl-Legion-15095.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://zzl-Legion:35707/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/zzl/RL/gym-...
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 30.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /use_sim_time: True

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    gazebo (gazebo_ros/gzserver)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    spawn_turtlebot_model (gazebo_ros/spawn_model)

auto-starting new master
process[master]: started with pid [15122]
ROS_MASTER_URI=http://zzl-Legion:12861/

setting /run_id to 7f5d39dc-4343-11e9-9b7f-0f45a3b6c431
process[rosout-1]: started with pid [15146]
started core service [/rosout]
process[gazebo-2]: started with pid [15149]
process[spawn_turtlebot_model-3]: started with pid [15150]
process[mobile_base_nodelet_manager-4]: started with pid [15152]
process[cmd_vel_mux-5]: started with pid [15154]
process[bumper2pointcloud-6]: started with pid [15156]
process[robot_state_publisher-7]: started with pid [15157]
Gazebo multi-robot simulator, version 7.14.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1552229281.760245218]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1552229281.760432834]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.113
[ INFO] [1552229282.386031331, 2.770000000]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1552229282.414351273, 30.080000000]: Physics dynamic reconfigure ready.
[spawn_turtlebot_model-3] process has finished cleanly
log file: /home/zzl/.ros/log/7f5d39dc-4343-11e9-9b7f-0f45a3b6c431/spawn_turtlebot_model-3*.log
[ INFO] [1552229283.068702233, 158.390000000]: Camera Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1552229283.105265937, 158.390000000]: Camera Plugin (ns = /)  <tf_prefix_>, set to ""
[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
[ INFO] [1552229283.327382783, 158.390000000]: Laser Plugin (robotNamespace = /), Info: Using the 'robotNamespace' param: '/'
[ INFO] [1552229283.327572883, 158.390000000]: Starting Laser Plugin (ns = /)!
[ INFO] [1552229283.347544095, 158.390000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1552229283.374344251, 158.390000000]: Starting plugin Kobuki(ns = //)!
[ WARN] [1552229283.374481743, 158.390000000]: Kobuki(ns = //): missing <rosDebugLevel> default is na
[ INFO] [1552229283.376501150, 158.390000000]: Kobuki(ns = //): <tf_prefix> = 
[Dbg] [gazebo_ros_kobuki.cpp:57] Plugin model name: mobile_base
[ INFO] [1552229283.376977982, 158.390000000]: Will publish tf. [mobile_base]
[ INFO] [1552229283.381126169, 158.390000000]: Kobuki(ns = //): Advertise joint_states[joint_states]!
[ INFO] [1552229283.382428469, 158.390000000]: Kobuki(ns = //): Advertise Odometry[odom]!
[ INFO] [1552229283.388575626, 158.390000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1552229283.394782718, 158.390000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/reset_odometry!
[ INFO] [1552229283.399864421, 158.390000000]: Kobuki(ns = //): Try to subscribe to mobile_base/commands/motor_power!
[ INFO] [1552229283.401429960, 158.390000000]: Kobuki(ns = //): Advertise Cliff[mobile_base/events/cliff]!
[ INFO] [1552229283.402417361, 158.390000000]: Kobuki(ns = //): Advertise Bumper[mobile_base/events/bumper]!
[ INFO] [1552229283.403196602, 158.390000000]: Kobuki(ns = //): Advertise IMU[mobile_base/sensors/imu_data]!
[ INFO] [1552229283.403231969, 158.390000000]: GazeboRosKobuki plugin ready to go! [mobile_base]
[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/pose/local/info, deleting message. This warning is printed only once.
[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[depth] not suppported.
[Wrn] [msgs.cc:1808] Conversion of sensor type[imu] not suppported.
```

nzlz, 2019-03-11:
Hi @zhaolongkzz ,

Even if imu and depth show warnings I believe the LIDAR plugin in charge of feeding the data to the neural network should not be affected. Can you launch the GUI (check readme) and post a .gif or video of the behavior you observe?

Also make sure setup_kinetic and turtlebot_setup do their job correctly and show no errors.

zhaolongkzz, 2019-03-11:
I upload a short video in google drive. [link](https://drive.google.com/open?id=1igXbAYim_MtvGy--Q-KJlopodzdMzoWz)
You can see the turtlebot is just spinning, it don't move to anywhere.

nzlz, 2019-03-11:
Seems like everything is fine except the obvious. You even get the LIDAR and the circuit loading fine. Can you check that turtlebot_setup.bash is executed correctly? Specially [these](https://github.com/erlerobot/gym-gazebo/blob/ba2058faaf39625a4e826ff3204ae76c658518b9/gym_gazebo/envs/installation/turtlebot_setup.bash#L10-L11) lines.

The spinning error is a problem with turtlebots inertias which is negligible when it is moving. You see it spinning because ROS communication is not working fine for some reason.