erlerobot/gym-gazebo #225: Installation error during "bash setup_melodic.bash". Invoking "make -j8 -l8" failed

anubhav1772, 2019-03-10:
Hi, I am a newbie in ros and gazibo. I'm trying to build the ros workspace on Ubuntu 18.04 but getting the following error. I am trying to fix it but it's not working. Your prompt help will be much appreciated. Thank you.

[ 78%] Building CXX object ros_control/controller_manager_tests/CMakeFiles/controller_manager_tests.dir/src/my_robot_hw.cpp.o
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::ProjPoints::AddPointsUsingChessboard(IplImage*, double, int, int, bool)’:
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:54:26: error: ‘cvFindChessboardCorners’ was not declared in this scope
  int pattern_was_found = cvFindChessboardCorners(gray, cvSize(etalon_rows, etalon_columns), corners, &point_count);
                          ^~~~~~~~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:73:3: error: ‘cvDrawChessboardCorners’ was not declared in this scope
   cvDrawChessboardCorners(image, cvSize(etalon_rows, etalon_columns),
   ^~~~~~~~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp: In member function ‘bool alvar::Camera::LoadCalibOpenCV(const char*)’:
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:2: error: ‘CvFileStorage’ was not declared in this scope
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
  ^~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:2: note: suggested alternative: ‘CvMemStorage’
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
  ^~~~~~~~~~~~~
  CvMemStorage
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:17: error: ‘fs’ was not declared in this scope
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                 ^~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:17: note: suggested alternative: ‘ffs’
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                 ^~
                 ffs
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:54: error: ‘CV_STORAGE_READ’ was not declared in this scope
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                                                      ^~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:54: note: suggested alternative: ‘CV_SHAPE_RECT’
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                                                      ^~~~~~~~~~~~~~~
                                                      CV_SHAPE_RECT
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:22: error: ‘cvOpenFileStorage’ was not declared in this scope
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                      ^~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:190:22: note: suggested alternative: ‘cvClearMemStorage’
  CvFileStorage* fs = cvOpenFileStorage(calibfile, 0, CV_STORAGE_READ);
                      ^~~~~~~~~~~~~~~~~
                      cvClearMemStorage
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:193:3: error: ‘CvFileNode’ was not declared in this scope
   CvFileNode* root_node = cvGetRootFileNode(fs);
   ^~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:193:3: note: suggested alternative: ‘TiXmlNode’
   CvFileNode* root_node = cvGetRootFileNode(fs);
   ^~~~~~~~~~
   TiXmlNode
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:193:15: error: ‘root_node’ was not declared in this scope
   CvFileNode* root_node = cvGetRootFileNode(fs);
               ^~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:193:27: error: ‘cvGetRootFileNode’ was not declared in this scope
   CvFileNode* root_node = cvGetRootFileNode(fs);
                           ^~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:193:27: note: suggested alternative: ‘cvNextTreeNode’
   CvFileNode* root_node = cvGetRootFileNode(fs);
                           ^~~~~~~~~~~~~~~~~
                           cvNextTreeNode
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:195:15: error: ‘intrinsic_mat_node’ was not declared in this scope
   CvFileNode* intrinsic_mat_node = cvGetFileNodeByName(fs, root_node, "intrinsic_matrix");
               ^~~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:195:36: error: ‘cvGetFileNodeByName’ was not declared in this scope
   CvFileNode* intrinsic_mat_node = cvGetFileNodeByName(fs, root_node, "intrinsic_matrix");
                                    ^~~~~~~~~~~~~~~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:196:51: error: ‘cvRead’ was not declared in this scope
   CvMat* intrinsic_mat = reinterpret_cast<CvMat*>(cvRead(fs, intrinsic_mat_node));
                                                   ^~~~~~
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:196:51: note: suggested alternative: ‘cvRemap’
   CvMat* intrinsic_mat = reinterpret_cast<CvMat*>(cvRead(fs, intrinsic_mat_node));
                                                   ^~~~~~
                                                   cvRemap
/home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/Camera.cpp:208:15: error: ‘dist_mat_node’ was not declared in this scope
[... 145 lines cut ...]
[ 78%] Built target ecl_geometry
[ 78%] Building CXX object ros_control/controller_manager_tests/CMakeFiles/controller_manager_tests.dir/src/vel_eff_controller.cpp.o
[ 78%] Linking CXX shared library /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/libecl_statistics.so
[ 78%] Linking CXX shared library /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/libcontroller_manager_tests.so
[ 78%] Built target controller_manager_tests
[ 78%] Built target ecl_statistics
In file included from /usr/include/pcl-1.8/pcl/sample_consensus/sac_model.h:52:0,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/sac.h:45,
                 from /usr/include/pcl-1.8/pcl/sample_consensus/ransac.h:44,
                 from /usr/include/pcl-1.8/pcl/registration/icp.h:45,
                 from /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/include/ar_track_alvar/filter/kinect_filtering.h:44,
                 from /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/src/ar_track_alvar/ar_track_alvar/src/kinect_filtering.cpp:36:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h: In function ‘void __static_initialization_and_destruction_0(int, int)’:
/usr/include/pcl-1.8/pcl/sample_consensus/model_types.h:99:3: warning: ‘pcl::SAC_SAMPLE_SIZE’ is deprecated: This map is deprecated and is kept only to prevent breaking existing user code. Starting from PCL 1.8.0 model sample size is a prot [... cut ...]
[ 78%] Linking CXX shared library /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/libkinect_filtering.so
[ 78%] Linking CXX shared library /home/anubhav1772/Documents/RL/gym-gazebo/gym_gazebo/envs/installation/catkin_ws/devel/lib/libgazebo_ros_triggered_camera.so
[ 78%] Built target gazebo_ros_triggered_camera
[ 78%] Built target kinect_filtering
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed
## ROS workspace compiled ##

nzlz, 2019-03-11:
Hi @anubhav1772 ,

Seems like OpenCv is not in your system, you could try : `sudo apt -y install python-opencv`. If you keep getting errors with the ar_track_alvar package, please post the issue [here](https://github.com/ros-perception/ar_track_alvar/issues).

anubhav1772, 2019-03-11:
Yeah, I installed ROS Desktop-Full version
Even after installing opencv using the above mentioned command, it's not working.